MKFirmware/MK4duo #456: Enstops ignored while homing

DvDriel, 2018-04-11:
On my delta setup, the enstops are being ignored while homing (G28). When using a G1 command to move the carriages all the way to the top the enstops do work and stop the movement. For G28 though the carriages try to push through the end. Any idea what might cause this issue?

I'm currently using MK4duo v3.3.5 on a delta setup with an Ultratronics board. Thanks in advance. With v3.3.3 I had the exact same issue.

Configuration_Delta.h:
#define ENDSTOPPULLUP_XMAX    true
#define ENDSTOPPULLUP_YMAX    true
#define ENDSTOPPULLUP_ZMAX    true
#define ENDSTOPPULLUP_ZPROBE  true

#define X_MAX_ENDSTOP_LOGIC   true   // set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_LOGIC   true   // set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_LOGIC   true   // set to true to invert the logic of the endstop.
#define Z_PROBE_ENDSTOP_LOGIC false   // set to true to invert the logic of the probe.

MagoKimbra, 2018-04-11:
Do you use EEPROM? This valor saved in eeprom...

DvDriel, 2018-04-18:
No,  I didn't enable EEPROM in the feature.h settings. Any thing else that could cause this problem?

MagoKimbra, 2018-04-18:
Check the endstop with M119

DvDriel, 2018-04-18:
That's what I did first. The endstops are working correctly. When all carriages are moved with a G1 command and touch the endStop they trigger and the movement is stoppped. The problem is with the homing command, which completly ignores the endStops.

DvDriel, 2018-05-09:
Any clue what this might be?

MagoKimbra, 2018-05-09:
This is impossible, the endstops are activated only during the home phase and not during the movement. So check that everything is OK with the M119 command.

DvDriel, 2018-06-15:
Still not able to fix it. Using the m119 command everything works fine, the end stops trigger as expected. Sending a g28 home all command though still does not work, the end stops are still ignored.

HansPrint, 2020-10-29:
Solution that worked for me:
Solder a 10k resistor between + and signal wires of the endstop. On my Ultratronics board, some of the endstop internal pull-up resistors were not active/broken. 
Regards,
Hans