MKFirmware/MK4duo #458: ADVANCED_PAUSE_FEATURE possible bug?

carlonb, 2018-04-12:
**Please, follow this template when opening a new issue: this will help us a lot in the process of helping you.**

### Issue description
When I use the command from LCD Prepare->Change Filament->Unload filament E1 (I've 4 extruders with the Flux capacitor) instead of retract the filament lenght, the extruder extract (put out) the filament.
The same happen if I use the gcode command m702.

### Firmware version
MK4duo 4.3.5 11/04/2018

My opinion is that in the file .....\MK4duo\src\core\commands\gcode\feature\m701_m702.h there is a bug in line 147:
`{
      // Unload length
      const float unload_length = -FABS(parser.seen('U') ? parser.value_axis_units(E_AXIS) :
                                                          filament_change_unload_length[tools.target_extruder]);

      unload_filament(unload_length, true, ADVANCED_PAUSE_MODE_UNLOAD_FILAMENT); 
    } `
In the statement       unload_filament(unload_length, true, ADVANCED_PAUSE_MODE_UNLOAD_FILAMENT); 
the variable unload_length must have the negative sign (-unload_length).
I've tested this and seems to work correctly.
Please consider that I do not deeply know the firmware, may be I'm wrong.
Bye

carlonb, 2018-05-20:
Hi all,
noboby commented,  am i wrong? If I'm wrong please someone suggest me a solution instead of modify the code by myself as above mentioned.
Grazie, Ciao

carlonb, 2018-08-16:
Any news?
Thanks.

carlonb, 2018-08-18:
Ok, I will wait for a ultimate solution.
Another question: Why changing the filament we need the intermediate "Quickly purge" as per the code above shown and confirmed with my tests ?
Thankyou

carlonb, 2018-09-04:
Is this issue solved in last 4.3.6?

iosonopersia, 2018-09-04:
@MagoKimbra guarda qui sopra, avevo scritto un messaggio in cui ho dimostrato che nel codice di m701_m702.h c'è un segno - di troppo... Volevo attendere il tuo parere prima di procedere, poi però mi sono dimenticato di questa issue, mi scuso ancora con Carlo

carlonb, 2018-09-04:
Non dovete scusarvi, devo solo ringraziarvi :)

iosonopersia, 2018-09-06:
@MagoKimbra , cosa ne pensi?

closed 2018-09-06 by iosonopersia

carlonb, 2018-09-09:
Tutto bene, ho caricato la 3.4.6 ed ora il retract del filamento funziona correttamente.