MKFirmware/MK4duo #465: Compilation error for Core XY and other

Hans007a, 2018-04-25:
Here the next error:

```
sketch\src\core\mechanics\core\core_mechanics.cpp:240:8: error: 'Cartesian_Mechanics' has not been declared

   bool Cartesian_Mechanics::prepare_move_to_destination_mech_specific() {

        ^

sketch\src\core\mechanics\core\core_mechanics.cpp: In function 'bool prepare_move_to_destination_mech_specific()':

sketch\src\core\mechanics\core\core_mechanics.cpp:259:15: error: 'current_position' was not declared in this scope

           if (current_position[X_AXIS] != destination[X_AXIS] || current_position[Y_AXIS] != destination[Y_AXIS]) {

               ^

sketch\src\core\mechanics\core\core_mechanics.cpp:259:43: error: 'destination' was not declared in this scope

           if (current_position[X_AXIS] != destination[X_AXIS] || current_position[Y_AXIS] != destination[Y_AXIS]) {

                                           ^

In file included from sketch\src\core\mechanics\core\../../../../MK4duo.h:62:0,

                 from sketch\src\core\mechanics\core\core_mechanics.cpp:29:

sketch\src\core\mechanics\core\core_mechanics.cpp:263:59: error: 'feedrate_mm_s' was not declared in this scope

               abl.bilinear_line_to_destination(MMS_SCALED(feedrate_mm_s));

                                                           ^

sketch\src\core\mechanics\core\../../../../src/inc/macros.h:217:28: note: in definition of macro 'MMS_SCALED'

 #define MMS_SCALED(MM_S) ((MM_S) * mechanics.feedrate_percentage * 0.01)

                            ^

sketch\src\core\mechanics\core\core_mechanics.cpp:271:36: error: 'feedrate_mm_s' was not declared in this scope

     line_to_destination(MMS_SCALED(feedrate_mm_s));

                                    ^

sketch\src\core\mechanics\core\../../../../src/inc/macros.h:217:28: note: in definition of macro 'MMS_SCALED'

 #define MMS_SCALED(MM_S) ((MM_S) * mechanics.feedrate_percentage * 0.01)

                            ^

sketch\src\core\mechanics\core\core_mechanics.cpp:271:50: error: 'line_to_destination' was not declared in this scope

     line_to_destination(MMS_SCALED(feedrate_mm_s));
```

and in the m420.h line 159 (158-160) it should be:
```
            #if ENABLED(ABL_BILINEAR_SUBDIVISION)
              abl.virt_interpolate();
            #endif
```
and not `bedlevel.virt_interpolate();`

Hans007a, 2018-04-26:
Compile works afer the patch, but the BIG BUG on CoreXY steps is not fixed.
Once again the bug description:
at homing (g28) is the step length in all directions (x, y and z) a half of the normal step length and the bed leveling is made not at the middle of the bed, but at the 1/4 in the left front position. 
with the last version is the abl (g29) ok!
I will try to print anything to look hw is there with the length of the steps...

My other cartesian printer works with all versions fully ok!

MagoKimbra, 2018-04-26:
Your CORE_FACTOR??

Hans007a, 2018-04-26:
#define CORE_FACTOR 1 for CoreXY

MagoKimbra, 2018-04-26:
ok, but there is no bug for the coreXY, others are fine. Sure of the steps for mm?

Hans007a, 2018-04-26:
yes! I can see it at the auto home all axis, but when I move the nozzle by hand all is ok...
I will try to print now to check the steps.

btw. at the auto home I can hear that the motors are goint slowlier or so...

Hans007a, 2018-04-26:
ok,the print is very bad! the second layer is displaced in the direction right-behing at the half of the object... very bad...

Hans007a, 2018-04-26:
my printer works well with the firstly 4.3.5 version

Hans007a, 2018-04-26:
I will now check the move the axis by the hand with a measuring stick.
1. the x-y position on the display are wrong after auto home
2. the x and y move are (i think) right for 10mm step
3. the z move is ~7mm for the 10mm step

Hans007a, 2018-04-26:
I will now write my backup version of 4.3.5 and check it again for test

Hans007a, 2018-04-27:
I uploaded my working source code and hope that You can find why this is working.
[MK4duo_X5S.zip](https://github.com/MKFirmware/MK4duo/files/1956132/MK4duo_X5S.zip)

I can made a tiny video of the buggy auto homing, if You need and post it...

Hans007a, 2018-04-27:
it is wrong? this is working well with my cartesian printer

MagoKimbra, 2018-04-27:
In cartesian do you have arduino due??

Hans007a, 2018-04-27:
yes, with 3-1 nozzle

MagoKimbra, 2018-04-27:
Ok... I really do not understand where the problem is. I checking the core routines ...

Hans007a, 2018-04-27:
I think, it can be a time problem, because I checked the source code too, and the procedures for core are the same...

Hans007a, 2018-04-29:
it was MINIMUM_STEPPER_PULSE! I set it to 2 and now works :)

Thank You for this hint!!!

closed 2018-04-29 by Hans007a