MKFirmware/MK4duo #468: Bug: after press "stop print" nozzle move to park position

Hans007a, 2018-04-29:
here a new bug:

after press "stop print" nozzle move to park position (the z coord too) and that can colide with the printed item.
The nozzle should drive only up and then at the side, but never down!

Tested with cartesian printer with advanced pause feature enabled.

I test now what the nozzle is doing after a fully print

iosonopersia, 2018-04-29:
Are you using the nozzle park feature? As far as I know, to achieve the behaviour you want, the command to park the nozzle without  crashing with the object is "G27 P2". But this is useful only if you can stop the print sending gcodes manually with a host software. I actually don't know how it works when the print is stopped from the display...

Hans007a, 2018-04-29:
I print always from the sd card and the old versions (before the stop bug) was ok

Hans007a, 2018-08-07:
now its ok

closed 2018-08-07 by Hans007a