MKFirmware/MK4duo #471: Bug: Voltage on the Ruramps cooler

gogsi1, 2018-05-10:
### Description of the problem
On the cooler, the voltage does not change, it ignores the specified limit values pid max.
With my settings should be 6.2v.

### Firmware Version
4.3.6

Did you manually configure MK4duo? Then write here a list of parameters that can be associated with the problem:
#define COOLER_PID_MAX 64
#define COOLER_PID_DRIVE_MIN 30
#define COOLER_PID_DRIVE_MAX 64

MagoKimbra, 2018-05-10:
Do you have a water cooler on hotend??

gogsi1, 2018-05-10:
No, I have an ordinary cooler on 25x25x10 with 12v.
In mk4due 4.3.4 this option worked. Before, I could reduce tension.

MagoKimbra, 2018-05-10:
But this is a fan not cooler!!!

gogsi1, 2018-05-10:
sorry,it's my fault because i know English not very well

gogsi1, 2018-05-11:
I even confused the fan with the cooler, but the essence of the problem has not changed.

MagoKimbra, 2018-05-11:
In the cards there are fets controlled by a processor pin, from there or the 3.3v 0v with a PWM frequency, this frequency causes the fet to be on for a time t 'and off for a time t ", doing in so that the fan goes more or less fast, but the output of the fet always remains of the same voltage this can not be controlled.

closed 2018-08-04 by iosonopersia