MKFirmware/MK4duo #474: G28 X curious behavior by the sd print

Hans007a, 2018-05-13:
When I send G28 X or G28 X0 thought USB the nozzle move only to the X-end stop, but when this command is at the end of the gcode on the sd card then the nozzle move to X-end-stop and then going down to the Z = 4mm on my CoreXY printer with BLTouch. That is very strange and can destroy the printer or the printed thing...

I don't set park position (//#define NOZZLE_PARK_FEATURE), but #define Z_SAFE_HOMING at the middle of the bed...

Maybe its only a setting, but I don't know which one...

MagoKimbra, 2018-05-13:
Take the test and make a complete home. Then go up to a value you want and then give the command G28 X0 and explain to me exactly what it does ...

Hans007a, 2018-08-07:
working now

closed 2018-08-07 by Hans007a