MKFirmware/MK4duo #475: Issues: initialize eeprom and 5 button Anet A8 con original board Anet.

FreeLuck, 2018-05-15:
Hello,
I decided to switch from the Marlin firmware to the MK4Duo v4.3.6 for my Anet A8 with the 5-key keyboard, since I saw that now there is the possibility to activate SLIM_LCD_MENUS that allows you to insert new features given the little space in the motherboard Anet.
I have configured, compiled and uploaded but I have problems:
the 5 keys are very sensitive and makes it impossible to move around in the menu, also sometimes the down key performs the operations that make the central key.
I set also `#define ENCODER_PULSES_PER_STEP 1` 
`#define ENCODER_STEPS_PER_MENU_ITEM 1` but without success.
The second problem is that the eeprom is not initialized, sent M502 but nothing happens as well as M500, at every start it tells me "WARNING EEPROM Changed ...".
I tried version 4.3.4, with basic firmware with more features disabled due to the little space, same problem with the keys but the eeprom works.

This is my v4.3.6 configuration file
[Configuration_Overall.zip](https://github.com/MKFirmware/MK4duo/files/2006498/Configuration_Overall.zip)

### My setup
Anet A8 Bowden
Anet stock motherboard
Autolevel 3dTouch

Thank you, I await your reply.

iosonopersia, 2018-05-15:
For the EEPROM problem, try sending a M500 and then a M501. It should overwrite the previous EEPROM data which apparently was not written by a MK4duo 4.3.6 firmware. Every MK4duo version is able to read only the data generated by the same version of the firmware

MagoKimbra, 2018-05-16:
Please add in configuration_overall.h if you configure firmware with Configurator online, or add in configuration_feature.h
#define ADC_KEYPAD_DEBUG after #define ZONESTAR_LCD

And write me the valor when not botton on keypad pressed and valor for single botton pressed..

FreeLuck, 2018-05-16:
Ok @MagoKimbra, if all goes well, these are the values:

without pressing any key = 255

[image]

[image]

[image]

[image]

[image]

MagoKimbra, 2018-05-16:
Thank's.. Now fix it

MagoKimbra, 2018-05-16:
Try now..

MagoKimbra, 2018-05-16:
It's you that you gave me up and down, in fact I had to reverse the values, they were correct before ...

FreeLuck, 2018-05-16:
Yes I know, in fact I checked again and confirmed what I said before; now the value of the up and down buttons correspond with the code and the respective KEYPAD_UP and KEYPAD_DOWN, it is the encoderPosition that is inverted giving the final result opposite, without the possibility to choose with `#define REVERSE_ENCODER_DIRECTION` or `#define REVERSE_MENU_DIRECTION`

[image]

[image]