MKFirmware/MK4duo #480: Bug: CoreXY #define Z_SAFE_HOMING don't work

Hans007a, 2018-05-21:
Afer the last commit:
[](https://github.com/MKFirmware/MK4duo/commit/221e19f977fe3e593ee27440a4c97154bc51a39b)

the feature #define Z_SAFE_HOMING don't works on my CoreXY system.

Now I will test it on my cartesian printer

MagoKimbra, 2018-05-21:
Why not function? Explain!

Hans007a, 2018-05-21:
I don't know why... I compiled the new code, and now is homing at x=0, y=0 and not at the middle of the bed...

Hans007a, 2018-05-21:
I compiled the last commit with core fix, and now its randomize...
the nozzle move allways to the x=0, but one time it don't move to the y=0 and middle of the bed. second time move to x=0 and y=0, but not to the middleof the bed, and one time it's doing all ok... very strange...

Hans007a, 2018-05-23:
I checked the last commit a10c64d and here are the results:
1. auto home from lcd
my both printer do z-homing random, one time at the middle of the bed, one time at the point x=0 y=0, one time at the back of the bed
2. level bed from lcd
both printer starts good with homing z in the middle of the bed and then move throught all the points. I tested it 3 times with both printers....

MagoKimbra, 2018-05-23:
Have you any function setting? QUICK_HOME or another?

Hans007a, 2018-05-23:
after todays bugfixes (I fixed by myself this double dot and the other bug, but this not good) the ato home from lcd do this:
first time good x, y, z
second time good x, forget to move to y=0, go back at the safe homing offset,  good z
third time good x, y, z
fourth time good x, forget to move to y=0, go back at the safe homing offset,  good z
and so on...

no, quick home is off

MagoKimbra, 2018-05-23:
Very strange, the command is same G28 and routine is same for CORE e Cartesian...
Check this...

Hans007a, 2018-05-23:
I wanted to upload my setting hiles, but here come an error...

I tested auto mome with both printers and they both have now this bug

Hans007a, 2018-05-23:
I updated my description of the todays tests for You, that You better understand (I hope...).
Read mt 7th post here
My english is not so good...

Hans007a, 2018-05-23:
now it works, here my setting files

[MK4duo_cartesian.zip](https://github.com/MKFirmware/MK4duo/files/2032751/MK4duo_cartesian.zip)
[MK4duo_coreXY.zip](https://github.com/MKFirmware/MK4duo/files/2032752/MK4duo_coreXY.zip)

MagoKimbra, 2018-05-24:
Please try with last commit!

Hans007a, 2018-05-24:
I tested the new commit. the results:
1. cartesian looks good, all works normal (auto home tested 4 times, level bed 2 times)
2. coreXY is wrong, auto home first time at x=0,y=0, then the next 2 times ok, but level bed starts with auto home wrong at the back of the bed (y=max) and starts to leveling bed at the half of y coord middle of the bed, x was ok... after reset auto home ok, leveling bed ok and the 2 times auto home ok. it look like the y coord is not cleared from time to time...

Hans007a, 2018-05-25:
It looks good! thank You!

closed 2018-05-25 by Hans007a

MagoKimbra, 2018-05-25:
Ok... Thank's...