MKFirmware/MK4duo #488: CoreXY Sensorless Homing TMC 2130

smoki3, 2018-05-26:
I cant compile  Sensorless Homing on my CoreXY with TMC 2130 drivers

`Arduino: 1.6.13 (Mac OS X), Board: "Arduino Due (Programming Port)"

In file included from sketch/src/core/mechanics/mechanics.h:383:0,
                 from sketch/MK4duo.h:124,
                 from /Users/smoki3/Downloads/MK4duo-master/MK4duo/MK4duo.ino:302:
sketch/src/core/mechanics/core/core_mechanics.h:103:21: error: 'static void Core_Mechanics::sensorless_homing_per_axis(AxisEnum, bool)' cannot be declared
         static void sensorless_homing_per_axis(const AxisEnum axis, const bool enable=true) override;
                     ^
In file included from sketch/MK4duo.h:124:0,
                 from /Users/smoki3/Downloads/MK4duo-master/MK4duo/MK4duo.ino:302:
sketch/src/core/mechanics/mechanics.h:366:20: error:   since 'virtual void Mechanics::sensorless_homing_per_axis(AxisEnum, bool)' declared in base class
       virtual void sensorless_homing_per_axis(const AxisEnum axis, const bool enable=true);
                    ^
In file included from sketch/src/core/mechanics/mechanics.h:383:0,
                 from sketch/MK4duo.h:124,
                 from /Users/smoki3/Downloads/MK4duo-master/MK4duo/MK4duo.ino:302:
sketch/src/core/mechanics/core/core_mechanics.h:103:21: error: 'static void Core_Mechanics::sensorless_homing_per_axis(AxisEnum, bool)' marked override, but does not override
         static void sensorless_homing_per_axis(const AxisEnum axis, const bool enable=true) override;
                     ^
exit status 1
Fehler beim Kompilieren für das Board Arduino Due (Programming Port).

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.
`

MagoKimbra, 2018-05-26:
Try now please!

smoki3, 2018-05-26:
I will try it tomorrow

smoki3, 2018-05-27:
The compile errors are gone. But the homing is not working. The printer is not detecting the endstops. https://github.com/MarlinFirmware/Marlin/commit/a446433c87dc8b715ed5b9d98a34104f90b676b2

Is this integrated?

MagoKimbra, 2018-05-27:
Yes.. Just integrated...

smoki3, 2018-05-27:
hmm my printer is crashing when i try homing

MagoKimbra, 2018-05-27:
Please retry now...

smoki3, 2018-05-28:
The axis are still crashing.

closed 2018-07-11 by smoki3