MKFirmware/MK4duo #490: Compiplation error of 4.3.6

onekk, 2018-05-29:
during the compilation of the new 4.3.6 an error is raised because some variables were renamed in the define phase but not in the relevant code.

` error: 'JUNCTION_DEVIATION_ACCELERATION' was not declared in this scope
         vmax_junction_sqr = (JUNCTION_DEVIATION_ACCELERATION * JUNCTION_DEVIATION_MM * sin_theta_d2) / (1.0 - sin_theta_d2);
                              ^
sketch/src/core/planner/planner.cpp:2044:64: error: 'JUNCTION_DEVIATION_MM' was not declared in this scope
         vmax_junction_sqr = (JUNCTION_DEVIATION_ACCELERATION * JUNCTION_DEVIATION_MM * sin_theta_d2) / (1.0 - sin_theta_d2);
`
#define JUNCTION_DEVIATION_FACTOR 0.05
#define JUNCTION_DEVIATION_ACC_FACTOR 1000

but **JUNCTION_DEVIATION_ACCELERATION**  was renamed in **JUNCTION_DEVIATION_ACC_FACTOR**

and **JUNCTION_DEVIATION_MM**  was renamed in **JUNCTION_DEVIATION_FACTOR** 

by your b45a0b7f28ef4c58fafd165a9ff79b1b49ce7207

Regards

Carlo D.

MagoKimbra, 2018-05-29:
No in configuration_feature the name is
```
//#define JUNCTION_DEVIATION

#define JUNCTION_DEVIATION_MM 0.02            // (mm) Distance from real junction edge
#define JUNCTION_DEVIATION_ACCELERATION 1000  // (mm/s²) Maximum centripetal acceleration
//#define JUNCTION_DEVIATION_INCLUDE_E
```

onekk, 2018-05-29:
Ok problem solved.

Reusing **Configuration_Overall.h** is not error safe.

I have supposed that staying in the same version the define are retained, but obviously I was wrong.

closed 2018-05-29 by onekk