MKFirmware/MK4duo #493: Enhancement: new Menu tree

onekk, 2018-06-04:
Hello, after having used for some time MK4duo, i will propose a different menu tree.

Sorry for the italian tree but my menu is in italian so
- Guarda
- Stampa da SD
- Home Assi
- Home e ABL
- Raffredda
- Preriscalda PLA (Bed and HotEnd)
- Preriscalda ABS (Bed and HotEnd)
- Prepara
- Controllo

The main goal is to put the most used function in the first menu seen.

As most of us use the SD if SD is present "Print from SD" has to be in the first positions.

The useful function of "Raffredda (Cool Down?)" is hidden in the "Prepare Menu" and also the Preheat PLA ABS and GUM.

An alternative is to have in the "Prepare Menu" a different tree in which some useful function are switched, maybe the Preheat one and the "Raffredda (Cool Down?)"

More to come if the improvement is feasible.

Regards

Carlo D.

gearsawe, 2018-08-10:
I would agree SD being 1st up. Me I use temperature, fan, flow rate

iosonopersia, 2018-09-06:
Hi @onekk , look at this: https://github.com/MarlinFirmware/Marlin/issues/10317 it may be interesting.

closed 2019-03-29 by iosonopersia