MKFirmware/MK4duo #495: One autofan on multiple hotends

scott0122, 2018-06-10:
Currently MK4duo seems to support one autofan on one hotend only. 
So I modify these files to make one autofan on multiple hotends.

***fan.cpp***
  void Fan::SetAutoMonitored(const int8_t h_index) {
    if (WITHIN(h_index, 0, HOTENDS - 1) || h_index == 7)
      SBI(autoMonitored, (uint8_t)h_index);
    else if (h_index == 99)
      LOOP_HOTEND() {
    	SBI(autoMonitored, (uint8_t)h);
  	  }
    else
      autoMonitored = 0;
    spin();
  }

***Configuration_Overall.h***
#define AUTO_FAN {-1,99,-1,-1,-1,-1}

### Firmware version
4.3.6

MagoKimbra, 2018-06-10:
No scott with the command you can add the Hotend for auto fan.
Example youhave tree hotend ok in configuration put one hotend #define AUTO_FAN {-1,0,-1,-1,-1,-1}
Now the fan 1 is auto fan on hotend 0.
M106 P1 H1
M106 P1 H2
Now the fan 1 is auto fan on H0 H1 and H2.
M500 for save in eeprom

You must clear auto fan on H1
M106 P1 H-1  // clear auto fan
M106 P1 H0
M106 P1 H2
Now fan 1 is auto fan on H0 and H2

scott0122, 2018-06-10:
Ok, thanks!

closed 2018-07-19 by MagoKimbra