MKFirmware/MK4duo #497: Compilation issue - MK4duo 4.3.6, Arduino 1.8.0

micheleamerica, 2018-06-14:
### Issue description
Revision 4.3.6 downloaded from Github on June 9 does not compile correctly giving several errors detailed below.

### Compile errors
C:\Users\Michele\AppData\Local\Temp\arduino_build_32249\sketch\src\core\stepper\stepper.cpp:1503:50: error: 'struct block_t' has no member named 'mix_steps'

           advance_dividend_m[i] = current_block->mix_steps[i] << 1;

In file included from C:\Users\Michele\AppData\Local\Temp\arduino_build_32249\sketch\src\core\stepper\../../../src/core/stepper/stepper.h:46:0,

                 from C:\Users\Michele\AppData\Local\Temp\arduino_build_32249\sketch\src\core\stepper\../../../MK4duo.h:133,

                 from C:\Users\Michele\AppData\Local\Temp\arduino_build_32249\sketch\src\core\stepper\stepper.cpp:79:

C:\Users\Michele\AppData\Local\Temp\arduino_build_32249\sketch\src\core\stepper\stepper.cpp: In static member function 'static void Stepper::set_directions()':

C:\Users\Michele\AppData\Local\Temp\arduino_build_32249\sketch\src\core\stepper\stepper.cpp:1896:19: error: 'active_extruder_driver' was not declared in this scope

         REV_E_DIR(active_extruder_driver);

C:\Users\Michele\AppData\Local\Temp\arduino_build_32249\sketch\src\core\stepper\../../../src/core/stepper/stepper_indirection.h:568:41: note: in definition of macro 'REV_E_DIR'

     #define    REV_E_DIR(E)     do{ if (E == 0) { E0_DIR_WRITE( isStepDir(E0_AXIS)); } else { E1_DIR_WRITE( isStepDir(E1_AXIS)); } }while(0)

C:\Users\Michele\AppData\Local\Temp\arduino_build_32249\sketch\src\core\stepper\stepper.cpp:1900:20: error: 'active_extruder_driver' was not declared in this scope

         NORM_E_DIR(active_extruder_driver);

C:\Users\Michele\AppData\Local\Temp\arduino_build_32249\sketch\src\core\stepper\../../../src/core/stepper/stepper_indirection.h:567:41: note: in definition of macro 'NORM_E_DIR'

     #define   NORM_E_DIR(E)     do{ if (E == 0) { E0_DIR_WRITE(!isStepDir(E0_AXIS)); } else { E1_DIR_WRITE(!isStepDir(E1_AXIS)); } }while(0)

### Firmware version
MK4duo 4.3.6 from June 9 2018

MagoKimbra, 2018-06-14:
Ok than's.. This evening fix it.

micheleamerica, 2018-06-14:
You are most welcome. You have here a fantastic piece of software.
Grazie mille

commit d1a75e27 (2018-06-14): Fix #497
closed 2018-06-14 by MagoKimbra