MKFirmware/MK4duo #503: Not enough memory for global variables

DotBreno, 2018-06-21:
### Issue description
Arduino IDE is not compiling firmware. It says it uses too much global variables

### Compile errors

Archiving built core (caching) in: C:\Users\craft\AppData\Local\Temp\arduino_cache_515111\core\core_arduino_avr_mega_cpu_atmega2560_0c812875ac70eb4a9b385d8fb077f54c.a
Sketch uses 123034 bytes (48%) of program storage space. Maximum is 253952 bytes.

**Global variables use 8562 bytes (104%) of dynamic memory, leaving -370 bytes for local variables. Maximum is 8192 bytes.**
Not enough memory; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing your footprint.
Error compiling for board Arduino/Genuino Mega or Mega 2560.

### Firmware version

FIRMWARE_NAME             "MK4duo"
SHORT_BUILD_VERSION       "4.3.6"
FIRMWARE_REVISION         "19062018"
BUILD_VERSION             FIRMWARE_NAME "_" SHORT_BUILD_VERSION
STRING_DISTRIBUTION_DATE  __DATE__ " " __TIME__    // build date and time
FIRMWARE_URL              "marlinkimbra.it"

### My setup
I use a RUMBA board with DHT11 for ambient temperature and humidity, Nextion 3.5" as display, and an ESP8266 hooked to serial3 pins. I used the configurator.

[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/2125298/Configuration_Pins.txt)

[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/2125295/Configuration_Overall.txt)

MagoKimbra, 2018-06-21:
Too many things that consume memory ... Unfortunately you have to remove something, the display for example. The nextion occupies a lot of ram.

iosonopersia, 2018-06-21:
You're using SD card, Nextion and Esp8266 to communicate with the printer... Choose one... Esp8266 is already an host, maybe you can avoid the sdsupport which requires a lot of ram...

closed 2018-07-19 by MagoKimbra