MKFirmware/MK4duo #510: Core Mechanism not working as expected

Andy-27, 2018-06-27:
### Issue description
I set up the firmware in config files as COREYX mechanism. My machine is running with RADDS 1.6 Board with RAPS driver and original Arduino Due. The 2 Nema17 Steppers for Core mechanism are connected to X and Y drivers. In principal everything is running fine just one axis goes wrong direction and I tried all different cases for invert stepper direction as well change core mechanism from COREYX to COREXY. Attached table shows the result when giving direction on certain axis to move, e.g. X move to +, or homing, if the axis is doing this correct. Unfortunately I can't get it run properly.

[image]

It looks like that first setup (Invert-Dir-X=false, Invert-Dir-Y=false) as well as third setup (Invert-Dir-X=true, Invert-Dir-Y=false) results in same outcome which I would not expect. The "green" higligthed cells reflect right move, the "red" ones the wrong move. Only the variant for X and Y in green is missing  ;-)

I would appreciate if someone in the community has an idea where to look into or what need to be changed to run properly.

Many thanks in advance
 
### Firmware version
4.3.6 download 25th June 2018

MagoKimbra, 2018-06-27:
You know that the direction of the steppers is stored in eerpom and you can change it with the gcode commands, if you use the eeprom and change from configuration you will not change anything ...

Andy-27, 2018-06-27:
Thanks for your comments.
I did not activate eeprom. I changed different stepper dir inverts in config file and re-flashed firmware to board. Since the outcome is different it was changed but as table shows not in all of the 4 variants.

MagoKimbra, 2018-06-27:
Ok one question? The motor is mounted front you or rear printer?

Andy-27, 2018-06-27:
It is the classical core-xy design
[image]

Andy-27, 2018-06-27:
Right motor is connected to X driver and left stepper to Y driver, but also changed this. Then the “move X...” in above table is going to “move Y...”.

MagoKimbra, 2018-06-27:
The right configuration is by looking at the printer on the left X and on the right Y, if they are on the front as well as on the drawing.
Now please use command M569 for change dir and not configuration.
M569 X0 or M569 X1

Andy-27, 2018-06-28:
Grazie Mille - That was it, I changed the stepper motors, left to X and right to Y and by seting invert stepper direction on both to "true" with M569 everything work fine now.

MagoKimbra, 2018-06-28:
Prego!! You are welcome!

closed 2018-06-29 by Andy-27