MKFirmware/MK4duo #514: Enhancement: Automatic door to print chamber

maukcc, 2018-07-02:
We have an automated door to the print chamber.
When printing starts, door closes automatically, when print finishes, the door opens automatically.
At this moment we control the door steppermotors as an extruder output.
It would be nice to control it as a D(oor)-axis with a end stop (DOOR_OPEN_PIN).

MagoKimbra, 2018-07-02:
Then DOOR_OPEN_PIN is used to block printing when the door is open.
The easiest thing is to use a servo to make the door open.
Just add 1 servo, if you already have a servo as a probe, add another, and with the command M280 P [id servo] S [degrees] open and close it.
For example if you have an open and only servo it is 120 degrees of the servo and closed it is 10 degrees of the servo you put:
M280 P0 S10; To close
M280 P0 S120; To open.

maukcc, 2018-07-02:
The thing is that it is a sliding door with a travel of 550mm. So a servo is probably not going to cut it :)
could I use DUAL_X_CARRIAGE for this?

MagoKimbra, 2018-07-02:
No not possible.. The only solution then is the addition of an axis for door engine, but for now it is not expected.

iosonopersia, 2018-07-02:
Why not using an additional Arduino Nano or similar (very low cost) with a stepper module. You could search for an example code on the web to drive the stepper, without slowing down the main board computations

closed 2018-08-04 by iosonopersia