MKFirmware/MK4duo #517: Max inactivity time (30 minutes) Heaters switch off! while printing via serial(usb)

hantaski, 2018-07-05:
### Issue description
When printing using octoprint, after 30min, print is halted because of "cold extrusion prevented" which comes because all heaters are shut down in middle of the print. In octoprint's serial log, I can see line:

**2018-07-05 12:44:27,577 - Recv: Max inactivity time (30 minutes) Heaters switch off!**

after which all heaters are shut down. From the code, it looks like it is only checking if printing from SD card is active, not if something is coming through serial.

small snippet of serial log, where this happens:

2018-07-05 12:44:26,054 - Recv:  T:210.05 /210 B:58.94 /60 B@:0 @:134
2018-07-05 12:44:26,282 - Recv: ok
2018-07-05 12:44:26,288 - Send: N26427 G1 X137.606 Y104.751 E27.42879*81
2018-07-05 12:44:26,308 - Recv: ok
2018-07-05 12:44:26,313 - Send: N26428 G1 X137.825 Y105.121 E27.44372*87
2018-07-05 12:44:26,597 - Recv: ok
2018-07-05 12:44:26,602 - Send: N26429 G1 X134.698 Y108.248 E27.59708*89
2018-07-05 12:44:26,629 - Recv: ok
2018-07-05 12:44:26,633 - Send: N26430 G1 X135.288 Y108.248 E27.61753*80
2018-07-05 12:44:26,928 - Recv: ok
2018-07-05 12:44:26,933 - Send: N26431 G1 X138.045 Y105.491 E27.75274*85
2018-07-05 12:44:26,953 - Recv: ok
2018-07-05 12:44:26,959 - Send: N26432 G1 X138.264 Y105.861 E27.76766*81
2018-07-05 12:44:27,054 - Recv:  **T:210.05 /210 B:58.83 /60** B@:0 @:143
2018-07-05 12:44:27,253 - Recv: ok
2018-07-05 12:44:27,258 - Send: N26433 G1 X135.877 Y108.248 E27.88471*93
2018-07-05 12:44:27,285 - Recv: ok
2018-07-05 12:44:27,290 - Send: N26434 G1 X136.467 Y108.248 E27.90516*93
2018-07-05 12:44:27,568 - Recv: ok
2018-07-05 12:44:27,573 - Send: N26435 G1 X138.484 Y106.232 E28.00406*90
2018-07-05 12:44:27,577 - Recv: **Max inactivity time (30 minutes) Heaters switch off!**
2018-07-05 12:44:27,593 - Recv: ok
2018-07-05 12:44:27,598 - Send: N26436 G1 X138.704 Y106.602 E28.01898*95
2018-07-05 12:44:27,858 - Recv: ok
2018-07-05 12:44:27,863 - Send: N26437 G1 X137.057 Y108.248 E28.09973*88
2018-07-05 12:44:27,890 - Recv: ok
2018-07-05 12:44:27,895 - Send: N26438 G1 X137.647 Y108.248 E28.12017*81
2018-07-05 12:44:28,054 - Recv:  **T:210.05 /0 B:58.83 /0 B@:0** @:0
2018-07-05 12:44:28,137 - Recv: ok
2018-07-05 12:44:28,142 - Send: N26439 G1 X138.714 Y107.181 E28.17251*84
2018-07-05 12:44:28,163 - Recv: ok
2018-07-05 12:44:28,168 - Send: N26440 G1 X138.714 Y107.771 E28.19296*86
2018-07-05 12:44:28,393 - Recv: ok
2018-07-05 12:44:28,398 - Send: N26441 G1 X138.053 Y108.432 E28.22537*92
2018-07-05 12:44:28,428 - Recv: ok
2018-07-05 12:44:28,434 - Send: N26442 G1 X138.938 Y107.816 F7800.000*89
2018-07-05 12:44:28,637 - Recv: ok
2018-07-05 12:44:28,642 - Send: N26443 G1 F3600*76
2018-07-05 12:44:28,663 - Recv: ok
2018-07-05 12:44:28,668 - Send: N26444 G1 X139.377 Y107.378 E28.24636*95
2018-07-05 12:44:28,858 - Recv: ok
2018-07-05 12:44:28,863 - Send: N26445 G1 E25.24636 F5400.00000*41
2018-07-05 12:44:28,890 - Recv: ok
2018-07-05 12:44:28,894 - Send: N26446 G92 E0*69
2018-07-05 12:44:29,054 - Recv:  T:210.05 /0 B:58.68 /0 B@:0 @:0

### Compile errors

### Firmware version
4.3.6

### My setup
HEVO, arduino due&RADDS

MagoKimbra, 2018-07-05:
Please read the instructions GCODE commands. If you print from the host to tell the printer you are printing, you must use the M530 S1 command and to stop printing M530 S0. If the printer is not printing after 30 minutes all the heaters are turned off.

hantaski, 2018-07-05:
Ah.. ok.. Didn't knew that. Thanks. I will try that. It's just not easy to find these kind of info as almost everything is in italian. When I googled that "max inactivity.." thing, I only got one hit and it was source code. So sorry about this :)

closed 2018-07-05 by hantaski

hantaski, 2018-07-05:
Just and idea, should the fw detect that you are printing, if it's getting g codes? I didn't need to tell that with basic marlin.

MagoKimbra, 2018-07-05:
Ok and if I'm doing the tests sending gcode codes, there is no difference between those random data that those data to always print codes are.
In addition, the M530 command does many other things, the subcommand s1 puts the printer in print mode, from that moment start the print timers, the control of the wire end etc etc etc. It is a command that automatically repetier hosts if it recognizes repetier firmware, so it does not give it to others, it must be put in the start gcode. Marlin does not have this function ...

MagoKimbra, 2018-07-05:
Your reasoning is correct, but you have to look at it from the point of view of a programmer, this means checking every time a command arrives, like G1, you reset the timer. These are useless machine cycles, every time a check and since during the printing 99% of the commands is G1, you would have an increase in machine cycles to make the control and reset of the timer not indifferent. Instead, with the G530 S1 command, the timer is suspended, no control, no extra instructions to execute.