MKFirmware/MK4duo #520: RUMBA not booting with reprapdiscount smart controller LCD

IvanLev, 2018-07-09:
### Issue description
Firmware doesn't want to work on RUMBA board with a reprapdiscount smart controller LCD enabled. When I disable the LCD it works fine. It is not the connectors as I've rotated them and the LCD lights up.
After the first flash the LCD showed the boot screen once but stopped on it. Rebooting or reflashing didn't help and the lcd didn't show anything after this time. The boot stops at the line:
Echo:Free Memory: 3210 PlannerBufferBytes: 1456

I think that there is something wrong with the pins file as normal marlin didn't work too and blocked the uart so I needed to reflash the bootloader to change firmware after marlin.

### Firmware version
MK4duo Firmware Version V4.3.6

### My setup
In this configuration I have tmc2130 enabled but disabling them didn't help.
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/2178209/Configuration_Overall.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/2178210/Configuration_Pins.txt)

IvanLev, 2018-07-10:
Tried now installing 4.3.5. It boots but doesn't respont to commands and the screen doesn't work.
And I've been able to reproduce the boot screen. Just left 4.3.6 running for some minutes. Nothing new in the serial port

MagoKimbra, 2018-08-05:
The I2c serial port provides an addressing, to use two or more peripherals connected to it. But if the peripheral device has the defaul address, no more than two can be attacked. So unfortunately it is not possible to put both the eeprom and the display on the same port. They should first be configured by giving them different IDs and then modify the firmware to use the two IDs inserted and not the default one.

IvanLev, 2018-08-05:
@MagoKimbra perhaps you posted this in the wrong issue? I'm not using the i2c for anything. Or do you mean that if I disable the internal eeprom everything will work fine?

MagoKimbra, 2018-08-05:
In theory, no, because however the eeprom is connected to the i2c and would respond when the display is interrogated. I do not know the i2c well, I know the SPI port well, but I know for sure that they need an ID, and the same id is always used in marlin and MK which is the default one. But I could be wrong.

IvanLev, 2018-08-05:
Rumba doesn't have an external eeprom and the i2c isn't the source of the issue. Can I enable any debugging options to see what goes wrong during boot?

MagoKimbra, 2018-08-05:
Sorry but do not possible enabled debugging with arduino...
try with disabled and disconnect lcd if eeprom function.

IvanLev, 2018-08-05:
Everything works fine without the lcd enabled. Normal marlin works fine too. But if I enable the lcd it doesn't boot.