MKFirmware/MK4duo #522: Firmware configurator tool 4.3.6- Y Driver Type missing

jbaiao, 2018-07-13:
First of all, many congratulations on the redesign of the "Firmware configurator tool".

In the "Mechanics" tab, in the "Y axis stepper motor" fields, "Y Driver Type" is missing, identical to the "X Driver Type".
It is very useful for those who use the "TCM2130" drivers.

Tks!

jbaiao, 2018-07-13:
MagoKimbra, there is another insue.
In "Configuration_Motor_Driver.h" file, there is no option to put current for motors.

jbaiao, 2018-07-16:
@MagoKimbra, can you please check this?
Tks!

MagoKimbra, 2018-07-16:
when you use the coreXY the drivers must be the same for both axes so when you set the X axis with a TMC2130 also Y by default has the 2130.

jbaiao, 2018-07-17:
Yes, you are right.

But where do i change the value of "current for motors"?
There is no opcion... and if i put values in "Configuration_Motor_Driver.h", after send firmware, the values is always the same... 1000

Edit:
I see that you put this in "Configuration_Overall.h", but in configurator tool there is no options for that.

Configuration_Overall.h 
#define X_CURRENT 1000
#define X2_CURRENT 800
#define Y_CURRENT 1000
#define Y2_CURRENT 800
#define Z_CURRENT 1000
#define Z2_CURRENT 800
#define E0_CURRENT 1000
#define E1_CURRENT 1000
#define E2_CURRENT 1000
#define E3_CURRENT 1000
#define E4_CURRENT 1000
#define E5_CURRENT 1000
Can you make possible to have this option in "Firmware configurator tool "?

MagoKimbra, 2018-07-17:
Soon!!

closed 2018-07-19 by MagoKimbra