MKFirmware/MK4duo #524: BED_PID_MAX not work

robertoscipione, 2018-07-16:
**Please, follow this template when opening a new issue: this will help us a lot in the process of helping you.**

### Issue description
the parameter BED_PID_MAX not work

### Firmware version
Mk4duo 4.3.6

### My setup
/**
 * MK4duo Firmware for 3D Printer, Laser and CNC
 *
 * Based on Marlin, Sprinter and grbl
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 * Copyright (C) 2013 Alberto Cotronei @MagoKimbra
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program. If not, see <http://www.gnu.org/licenses/>.
 */

/**
 * Configuration_Overall.h
 * Here you can define all your custom settings and they will overwrite configurations in the main configuration files.
 */

/*******************************
 *   Firmware Version V4.3.6   *
 *******************************/

#define CONFIGURATION_OVERALL

/***********************
 * Configuration_Basic *
 ***********************/
#define SERIAL_PORT 0
#define BAUDRATE 250000
#define STRING_CONFIG_H_AUTHOR "(none, default config)"
#define MACHINE_UUID "6e8f83e2-c930-0417-c855-67227dc35639"
#define KILL_METHOD 0
#define NO_TIMEOUTS 1000
//#define ADVANCED_OK
//#define EMERGENCY_PARSER
//#define FASTER_GCODE_PARSER
//#define FASTER_GCODE_EXECUTE
#define HOST_KEEPALIVE_FEATURE
#define DEFAULT_KEEPALIVE_INTERVAL 2
#define MOTHERBOARD BOARD_RAMPS_13_HFB
#define MECHANISM MECH_CARTESIAN
//#define MECHANISM MECH_COREXY
//#define MECHANISM MECH_COREYX
//#define MECHANISM MECH_COREXZ
//#define MECHANISM MECH_COREZX
//#define MECHANISM MECH_COREYZ
//#define MECHANISM MECH_COREZY
//#define MECHANISM MECH_DELTA
//#define MECHANISM MECH_MORGAN_SCARA
//#define MECHANISM MECH_MAKERARM_SCARA
//#define MECHANISM MECH_MUVE3D
#define POWER_SUPPLY 0
#define PS_DEFAULT_OFF false
[... 1499 lines cut ...]
"filrunoutpin5": "NoPin",
"laserpwrpin": "ORIG_LASER_PWR_PIN",
"laserpwmpin": "ORIG_LASER_PWM_PIN",
"laserperipheralspin": "NoPin",
"laserperipheralsstatuspin": "NoPin",
"cncrouterpin": "NoPin",
"powerconsumptionpin": "NoPin",
"doorpin": "NoPin",
"powercheckpin": "NoPin",
"caselightpin": "NoPin",
"rgbledRpin": "NoPin",
"rgbledGpin": "NoPin",
"rgbledBpin": "NoPin",
"rgbledWpin": "NoPin",
"neopixelpin": "NoPin",
"dhtdatapin": "NoPin",
"END_DATA": 0
}
========== End configuration string ==========
*/

Have you configured MK4duo manually? Then write here a list of the settings that may be related to the issue:
- BED_PID_MAX 80

i have tryed on other firmware marlin and repetier and it works, thanks for the support

MagoKimbra, 2018-07-16:
This valor is saved in to EEPROM check with command if you have this valor. M306 H-1

robertoscipione, 2018-07-16:
Ok thanks now it work. The parameter now are stored in EEPROM.

This is the procedure:
M306 H-1 show actual PID max for the bed
M306 H-1 C140 (140 is the power, you need to chose your parameter)
M500 to save the new parameter to EEPROM

Thanks for the support

robertoscipione, 2018-07-16:
i need other question, why PID max for the bed not listed in the gui of the eeprom parameters from Repetier Host ?

thanks

iosonopersia, 2018-07-16:
It's because its definition is missing in the file marlineeprom.xml which tells RepetierHost what are the values stored in eeprom. I'm currently working on that file to complete it, but it will take some time...

robertoscipione, 2018-07-16:
I'm designing a Cartesian 3D printer and I see that mk4duo works well. When I finished I will release open source with fw mk4duo;) thanks for now

iosonopersia, 2018-08-25:
@robertoscipione you can find the complete marlineeprom.xml file inside src/plugin Repetier folder of MK4duo. Use it to replace the marlineeprom.xml file contained inside C://Program Files/Repetier-Host/plugins/RRFirmwares/ . This will let you modify almost every aspect of your printer! Have fun!