MKFirmware/MK4duo #528: Temperatur doesn't reach target while printing

smoki3, 2018-07-17:
### Issue description
I use dual extruder on my machine.When I start my print I heat up both extruders to 245°C and both reaches the target temperature. Then While printing with Tool 1, I set Tool 0 to 165°C. After I changer to Tool 0 then I use M109 S245 T0 to get printing temperature. But then the Temperature will be never reached. It only reaches 235°C. 

### Firmware version
latest master release

smoki3, 2018-08-09:
Yes. But if I heat up directly with fan no problem. Also if I turn of the fan it doesn't fix it

gearsawe, 2018-08-09:
The reason I was asking if it reached the target with the fan off for the second heating cycle was this would indicate you need to do the PID tuning with the fan on. In the past I have had all sorts of PID tuning problems. sometime I need a fan other times none. or it would over heat when on the first layers because it was close to the bed at 100C then if it did get past that it would be under temp if it was a tall print. I wish this had dead-time temp control. you get a little more temperature fluctuation but it just works. It could be 110F in my garage or 50F and it would not have a problem.