MKFirmware/MK4duo #530: Compile error: (LCD_BED_LEVELING) Enabled

remspoor, 2018-07-17:
### Issue description
I'm trying to compile with option LCD_BED_LEVELING enabled and got a compile error.

### Compile errors
`/tmp/arduino_build_104965/sketch/src/lcd/ultralcd.cpp: In function 'void lcd_prepare_menu()':
/tmp/arduino_build_104965/sketch/src/lcd/ultralcd.cpp:2576:45: error: '_lcd_ubl_level_bed' was not declared in this scope
       MENU_ITEM(submenu, MSG_UBL_LEVEL_BED, _lcd_ubl_level_bed);
                                             ^
/tmp/arduino_build_104965/sketch/src/lcd/ultralcd.cpp:283:30: note: in definition of macro '_MENU_ITEM_PART_2'
         menu_action_ ## TYPE(__VA_ARGS__); \
                              ^
/tmp/arduino_build_104965/sketch/src/lcd/ultralcd.cpp:297:39: note: in expansion of macro 'MENU_ITEM_P'
   #define MENU_ITEM(TYPE, LABEL, ...) MENU_ITEM_P(TYPE, PSTR(LABEL), ## __VA_ARGS__)
                                       ^
/tmp/arduino_build_104965/sketch/src/lcd/ultralcd.cpp:2576:7: note: in expansion of macro 'MENU_ITEM'
       MENU_ITEM(submenu, MSG_UBL_LEVEL_BED, _lcd_ubl_level_bed);
       ^`
 
### Firmware version
Git version V4.3.6 commit: d89c0f218b37edc1f832a0cc7a7365580e14a2ce

### My setup
I use an Ultratronics board on a self built CoreXY with the following configuration files:
[Configuration_Overall.h.txt](https://github.com/MKFirmware/MK4duo/files/2203498/Configuration_Overall.h.txt)
[Configuration_Pins.h.txt](https://github.com/MKFirmware/MK4duo/files/2203499/Configuration_Pins.h.txt)
There are a couple of settings changed after using the configurator, but i don't think it is relevant concerning this compile error.

remspoor, 2018-07-19:
I've seen de comments from bug #515. If i understand correctly, when using UBL, the bed leveling option is not needed. Should i close this bug then? Or should we keep this bug report open as a reminder for something?

MagoKimbra, 2018-07-19:
This isn't bug!! The LCD_BED_LEVEL not needed with UBL...

closed 2018-07-19 by MagoKimbra