MKFirmware/MK4duo #532: nextion 3.5" settings menu

weed2all, 2018-07-20:
i have installed the mk4duo_3_5_basic.tft from nextion fw, and all work great but i have a question about the settings menu, when click on settings  new page open with 2 new button the filament and temperature button, but when i click on temperature button nothing happen...does not this button supposed to have the preahet settings for pla and abs?

jbaiao, 2018-07-23:
@MagoKimbra, can you see this?
Tks!

iworld4us, 2018-07-23:
i will test it tomorrow

iworld4us, 2018-07-23:
https://github.com/iworld4us/nextion-2.4

weed2all, 2018-07-23:
Ok...thank you...

iworld4us, 2018-07-24:
Temperature button is available only in 5" and 7" screen.
In other sizes it does nothing.
If you want it, you can try to export pages (temp, setup, usertemp) from 5" project, import it and resize.

weed2all, 2018-07-24:
And how to apply to marlin code?

iworld4us, 2018-07-24:
nextion editor and compiler  for .HMI files: 

https://nextion.itead.cc/resources/download/nextion-editor/

weed2all, 2018-07-25:
Yes I'm aware of nextion editor...and I could work to port temperature function from 5" to 3.5" and 4.3" but how to apply that to marlin firmware in order to act like is supposed to?or the function is already in marlin firmware and only need to create the new page for temperature for my LCD?

iworld4us, 2018-07-25:
In Marlin the screen size influences only the graph size. Other functions work on any screen size.

weed2all, 2018-07-25:
Ohh...ok...thank you for your quick replay!ok I will try to export as much as I can from 5" to 3.5" and 4.3" as my buddy @Jbaiao need it as well

weed2all, 2018-07-25:
can you upload the .HMI file for 3.5" lcd pls

iworld4us, 2018-07-25:
[mk4duo_3_5_inch_v1_0_5.HMI.zip](https://github.com/MKFirmware/MK4duo/files/2227244/mk4duo_3_5_inch_v1_0_5.HMI.zip)

weed2all, 2018-07-25:
Thank you so much

weed2all, 2018-07-25:
it is empty sketch...nothing appears when open with nextion editor

iworld4us, 2018-07-25:
check your unzip !

iworld4us, 2018-07-25:
12.9mb

weed2all, 2018-07-25:
Yup...7zip no good...i get linear...now is ok...the sketch is open on nextion editor!

jbaiao, 2018-08-01:
It already works.
Only problem is that temperatures are defined by variables in the firmware of the netion ...
Otherwise it works very well :)

https://www.youtube.com/watch?v=d6arKKHOJvk

closed 2019-03-20 by weed2all