MKFirmware/MK4duo #534: Cooler target temperature at boot time

Micilera, 2018-07-21:
The cooler target temperature should not be set to zero at boot/init time, because in this way the cooler starts immediately.
Thanks

MagoKimbra, 2018-07-21:
The cooler is liquid or other cooling for the laser

Micilera, 2018-07-21:
For sure the cooler is liquid, but if the initial target temperature is set to zero, the refrigerator (or the device that freeze the liquid) starts immediately. I would propose to set the initial target temperature of the cooler to 35.

gearsawe, 2018-08-05:
use an auto run gcode at boot to set the them(would require SD card reader)? Better yet a string of gcode that could be store to eeprom which could run at start up. Easy for user to use this for ever they want? Then there would not be bunch of "preset" in the code. Or would this take up to much storage?

Micilera, 2018-09-05:
This issue has nothing to do with the laser. The cooler  can be used for cooling the water of a 3d printer