MKFirmware/MK4duo #535: Filament and flowmeter pins

Micilera, 2018-07-21:
The definition of the filament runout pin and the flowmeter pin should be in the board file, like:

#define ORIG_FIL_RUNOUT0_PIN    24
#define ORIG_FLOWMETER_PIN      26  

and in configuration_pin.h there should be a reference like:

  #define FIL_RUNOUT0_PIN     ORIG_FIL_RUNOUT0_PIN
...
  #define FLOWMETER_PIN ORIG_FLOWMETER_PIN

this is very similar to the handling of all the other pins.

MagoKimbra, 2018-07-21:
The ORIG pins are those dedicated by the board for that function, no board has those pins already defined, so you can not use ORIG.

Micilera, 2018-07-21:
I can understand that. But in this way,  at every sw update, I need to change your configuration_pin.h with my setting. I do not think this is the best way. What about the BED, CHAMBER and COOLER pins, are those already defined for each board? No I suppose, but their definition is in the board file.  What do you think about?

Micilera, 2018-09-05:
Brut it should  be great if you do this.