MKFirmware/MK4duo #539: Bug in bed level

Hans007a, 2018-07-29:
Because UBL dont works, I test the other BL and here the first bug:

```
sketch\src\core\commands\gcode/bedlevel/m420.h:160:15: error: 'abl' was not declared in this scope

               abl.virt_interpolate();

```
Regards

Hans

Hans007a, 2018-07-29:
Now its strange, my printer shous the splash screen and this was all...

MagoKimbra, 2018-07-29:
Why UBL not function?
The errore in abl is very strange...

Hans007a, 2018-07-29:
I dont know, why UBL doesnt work. Configurating the UBL works without problems, but not when printing. The z-axis dont move at all it doesnt matter how much the UBL is changed.
Another thing not working is the editing of the UBL points. 
After Hitting "Edit-Mesh" on the LCD-display it says, that the printer is going to home all the axis, but it doesnt to anything and the whole printer is  no longer responding. Resetting the EEPROM didnt change the result.