MKFirmware/MK4duo #540: PID settings SHOULD BE SAVED ALWAYS

Hans007a, 2018-07-29:
I have my PIS settings and I set it always in the configs.

Please disable this "please set PID" that is comming after each update!

MagoKimbra, 2018-07-29:
No problem you use M303 P I D with valor and m500, and you will not have the message anymore.

Hans007a, 2018-07-29:
Is this not the same from display:
1. CONTROL -> TEMPERATURE -> PID Autotune 
2. CONTROL -> Save Settings
because I do this each time...

MagoKimbra, 2018-07-29:
It is the same, but if you do not want to redo the autotune every time it is enough to enter the values with M301 P I D

Hans007a, 2018-07-30:
I find a bug on my cartesian printer. the firmware need a BED PID, but I have disabled this feature and the old version (16.07.2018) works well... very strange...

iosonopersia, 2019-03-29:
@Hans007a 

> I find a bug on my cartesian printer. the firmware need a BED PID, but I have disabled this feature and the old version (16.07.2018) works well... very strange...

Do you still get this error? Try with the latest version. If you encounter the same issue, please consider opening a new issue. Closing this one.

closed 2019-03-29 by iosonopersia