MKFirmware/MK4duo #543: Ignoring Extruder Feedrate on start

smoki3, 2018-08-01:
CoreXY 
RADDS 

### Issue description
The printer is ignoring the feedrates for the extruders on start or when i send a command over the display. As soon my printer is booted and a send an extrusion command via octoprint the extruder is moving extremely fast and loosing steps. As soon as I send the same command again then its working.

When i send a manual extrusion command via the display the extruder is always moving extremely fast. Als when i use the filament load and unload feature.

smoki3, 2018-08-06:
Is it possible that the printer is ignoring the Extruder feedrate when no  axis movement is in the command?

smoki3, 2018-08-06:
Is it possible to grab logs?

iosonopersia, 2018-08-06:
The printer knows the max feedrate of each axis (which you inserted during configuration or modified with M203). You can check what value is now used by your printer sending a M503 command (look for the line which starts with M203).

If the value is correct, then you should try to understand what exact command is sent by Octoprint to the printer (just look at the Terminal tab after hitting the "extrude" button). It should be something like this: `G1 Exxx Fyyy`. xxx is the position to which the extruder must go, yyy is the feedrate in mm/min. Maybe Octoprint is told by its config options to use a too high feedrate for manual moves. We need to know the exact command that is sent by Octoprint, before trying to discover what's the problem with your machine (if there's one).

smoki3, 2018-08-06:
Okay I will try later. It wired that it only happens after booting the printer

Unfortunaly it also happens when I use the filament load and unload feature (advanced parking feature).

smoki3, 2018-08-11:
Here the Eeprom stats:

```
Recv: Config:Maximum feedrates (units/s):
Recv: Config:  M203 X300.000 Y300.000 Z120.000
Recv: Config:  M203 T0 E100.000
Recv: Config:  M203 T1 E100.000
 ```

```
Recv: Config:Steps per unit:
Recv: Config:  M92 X160.000 Y160.000 Z1600.000
Recv: Config:  M92 T0 E302.000
Recv: Config:  M92 T1 E302.000
```
Output from Octoprint:

```
Send: G91
[...]
Send: G1 E5 F100
[...]
Send: G90
```
So i think here everthing looks fine

iosonopersia, 2018-08-11:
Well, if that's the command that generates the unwanted behavior, then the problem is not in the command itself but in how it's executed by the machine. Actually you have 100mm/s max speed and you're manually moving it at 100mm/min= 1,6667 mm/s, well below the max value.

We need your configuration. Are you using classic Jerk Control or Junction Deviation algorithm? Can you make a video so we can clearly see what's happening?

smoki3, 2018-08-11:
I use Junction Deviation. 
[config.zip](https://github.com/MKFirmware/MK4duo/files/2280281/config.zip)