MKFirmware/MK4duo #546: Bluetooth on latest version

pirat777knj, 2018-08-04:
Hi! 
I can't start Bluetooth on latest version(((
On older version (4.3.2) it working.
How do I start Bluetooth?

iosonopersia, 2018-08-05:
Oh, right! That's clear! So @pirat777knj please try selecting the serial port to which the bluetooth extension board is connected. Obviously, if you use bluetooth you can't use usb cable and viceversa... Let us know!

pirat777knj, 2018-08-06:
It's strange, it all worked) Although in the firmware I did not change anything.
But there was a strangeness: constantly timeout (
speed 57600

pirat777knj, 2018-08-08:
I'll add: the level of the table stopped working too (
team G28 works like this: home - X, home - Y, lowering Z and everything. The servo does not push the probe.
all switches are in good condition and configured the same way. what is the problem?

closed 2019-03-11 by pirat777knj

iosonopersia, 2019-03-11:
@pirat777knj , have you solved this issue?

pirat777knj, 2019-03-11:
Yes. Update to a newer version