MKFirmware/MK4duo #547: All Heater/Cooler Functions not work

gearsawe, 2018-08-05:
### Issue description
All Heater/Cooler Functions not working.
Board is Rumba, With LCD(SD, Click Encoder)
Works with Marlin 1.1.8 and Marlin Bugfix1.1.x
Set with dummy sensor table. Set Temp with controller above 25C. LCD indicates heating with dot on Extruder but light on board inactive (indication no power to output).
Bed, Chamber and Cooler for all the outputs are non-functional. Will note Fans do work. The "heater" based functions seem to be broken. Movement, Temp Sensors, End Stops work.

### Compile errors
sketch\src\core\eeprom\eeprom.cpp:2005:73: warning: missing terminating " character
CONFIG_MSG_START_E(DHT sensor parameters: P<Pin> S<type 11-21-22>:");
missing a quote after first open parenthesis. corrected and recompiled

### Firmware version
4.3.6

### My setup
online config used files attached. also tried manually with same results.
in attached file Configuration_Pin.h change following. Tried both with change and with the defaults from online configuration. These are the assigned pins on working marlin 1.1.8. but I assume pin assignment it not the root cause.
#define HEATER_0_PIN         2//ORIG_HEATER_0_PIN
#define HEATER_1_PIN         3//ORIG_HEATER_1_PIN
#define HEATER_BED_PIN       9//ORIG_HEATER_BED_PIN

[Configuration_Overall.h.txt](https://github.com/MKFirmware/MK4duo/files/2260092/Configuration_Overall.h.txt)
[Configuration_Pins.h.txt](https://github.com/MKFirmware/MK4duo/files/2260093/Configuration_Pins.h.txt)

gearsawe, 2018-08-09:
**Update**
all my testing has been done thru a the connected control panel. I finally hooked it up to Simplify3D once I set the temp for Extruder 1 the extruder starts to work and then I can set it via the connected panel.

here is what I observe.
-If I turn on the Extruder1 via Simplfy3D, I can then turn it off and on via control panel
-If I turn on the Extruder1 via Simplfy3D then back off, I can no longer turn it on and off via control panel.
-If I turn on the Extruder1 via Simplfy3D, I can then turn off by control panel with SD card I run a program and it turn on the extruder1  with M109.
-If I turn on the Extruder1 via Simplfy3D then back off. I can no longer turn it on via control panel or via gcode from SD card with M109.
-Also note if Extruder 1 via Simplfy3D. I cannot control Extruder 2 via control panel

So it seems if an Extruder, Heater, or Cooler is turned on Via Simplfy3D it can then be turned on or off Via a control panel. Once it is turned off Via Simplfy3D I can longer turn it on and off via control panel.
Also note even if the set temp is well below the current temp Via Simplfy3D where the Extruder does not turn on it can then be controled via the control panel

One last point. from SD I can set temps for heaters with M104, bed M140, cooler M142 then I can set via panel. If I use only 109, 190, 192 those gcodes with wait do not set heaters to active and can not control via controller. So when using controller it may not be setting heater to active? or making heater active when setting only using 109, 190...192.

gearsawe, 2018-08-20:
Looking at feature list from 4.3.5
"If the PID autotune is not done, the firmware will not switch the heater on until the autotune is performed. Only if have EEPROM."
I was using bang-bang. Is this feature causing the problem I am seeing here even though I am not using PID?

iosonopersia, 2018-08-25:
@gearsawe please try again with latest MK4duo. We've recently fixed a dangerous bug in the heaters code thanks to @Micilera suggestion #570

MagoKimbra, 2018-08-31:
The use of pid is saved in eeprom, sure to have the pid deactivated?
check with the M306 command if you are using pid or not. If you are using pid until the pid auto tune is done or eeprom pid is saved with the M301 command, the heaters are not turned on.

iosonopersia, 2018-09-04:
@gearsawe With https://github.com/MKFirmware/MK4duo/commit/603c86b98f58619cc6f6720b9430bb23fe06c0e3 and https://github.com/MKFirmware/MK4duo/commit/e4aee2cd9a2d0f473c82635916f5e193414d2bec , this issue should be solved. Try again with the latest version.

closed 2018-09-04 by iosonopersia