MKFirmware/MK4duo #552: Not compiled for Dual X-carriage

lis104, 2018-08-09:
### Issue description
Not compiled for Dual X-carriage

### Compile errors
Arduino 1.8.5

sketch\src\core\mechanics\mechanics.cpp: In static member function 'static float Mechanics::x_home_pos(int)':

sketch\src\core\mechanics\mechanics.cpp:476:14: error: 'base_home_pos' was not declared in this scope

       return base_home_pos[X_AXIS];

              ^

### Firmware version
MK4duo's version 4.3.6.

### My setup
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/2273121/Configuration_Overall.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/2273122/Configuration_Pins.txt)

iosonopersia, 2018-08-09:
@MagoKimbra In mechanics.cpp line 476  `base_home_pos` is used but it's defined only in subclasses core_mechanics.cpp and cartesian_mechanics.cpp . We can copy that function inside those classes but then we would have two identical copies of the same function to maintain. Or we can define it inside mechanics.cpp like in this example:
```cpp
#if IS_CARTESIAN || IS_CORE
    const float Mechanics::base_home_pos[XYZ]  = { X_HOME_POS, Y_HOME_POS, Z_HOME_POS };
#endif
```

lis104, 2018-08-09:
This variable is nowhere else in mechanics.cpp is not used - it is empty. Accordingly, when you return a value, we will get the wrong data.

lis104, 2018-08-09:
Not closed please.

iosonopersia, 2018-08-09:
@lis104 try compiling the new version, he closed this issue because he fixed this problem with a commit.

lis104, 2018-08-10:
@MagoKimbra thanks. I compiled - no errors. Try go on printer.

closed 2018-08-10 by lis104