MKFirmware/MK4duo #554: If DELTA_CARRIAGE_OFFSET is set to zero in web configurator, its value won't be set in Configuration_Overall.h

Bougakov, 2018-08-09:
### Issue description

If DELTA_CARRIAGE_OFFSET is set to zero in web configurator, then _#define DELTA_CARRIAGE_OFFSET_ in Configuration_Overall.h  won't be assigned any value

### Compile errors

expected primary-expression before ')' token

MagoKimbra, 2018-08-09:
Apart from the mistake, how do you get 0 this value? It is the distance between the rod point and the nozzle center.

Bougakov, 2018-08-11:
No. According to the picture in the wizard, https://marlinkimbra.it/configurator/images/deltamisure.jpg "Delta carriage offset" is "[4]", the distance between the tower and the pulley. I tried to zero it to simplify the equation.

MagoKimbra, 2018-08-11:
Yes, you're right, I confused 3 and 4, but the error would be the same. Both 3 and 4 can not be 0. Those values are important for calculating the delta radius which is the base of the triangle that has the diagonal rod.

Bougakov, 2018-08-11:
Duly noted; I am measuring my actual values right now to put them in the wizard

closed 2018-08-20 by iosonopersia