MKFirmware/MK4duo #558: Compile error "if (!forReplay)"

wesk-ms, 2018-08-13:
**Please, follow this template when opening a new issue: this will help us a lot in 

### Compile errors

Hello,

I downloaded a new build last night using the online configurator and I'm getting a compile error.  I'm not an home right now, so I can't copy and paste the error.  I can provide more information later though, if needed.

Going from memory:

Src/core/eeprom/eeprom.cpp
Line 2005
     if (!forReplay)

The 'forReplay' function hasn't been defined and I couldn't find that string anywhere else in the code, which explains the compilation error.

I have volumetric extrusion enabled.  I'm sure that disabling this setting would allow it to compile, but I thought volumetric extrusion was a good feature to enable.

Thanks,
Wes

Before posting this issue, however, take a long breath and check this two things:
1. If you're not using the latest version of Arduino IDE, please try again with the latest one.
2. Always check if an error (or more than one) starts with `DEPENDENCY ERROR: _something wrong!_`: **DO NOT POST DEPENDENCY ERRORS**, please! They are messages which tell you what config option was set badly. Fix your configuration and try again. If you also get other types of error, please be sure to have fixed dependency errors **before** posting this issue!

### Firmware version
MK4duo's versions that have this issue.

### My setup
If you used the online configurator, copy the content of `Configuration_Overall.h` and `Configuration_Pins.h` in two text files. **Then Drag&Drop them here. DO NOT COPY THE CONTENT DIRECTLY HERE!!!**

Have you configured MK4duo manually? Then write here a list of the settings that may be related to the issue:
- SETTING_CHANGED_1    true
- SETTING_CHANGED_2    25
- SETTING_CHANGED_3    "blablabla"
- ...

_If needed, you'll be asked for other settings you modified._

**Please be patient, we'll be trying to help you as soon as possible.**

iosonopersia, 2018-08-14:
@wesk-ms Thanks! Fixed with https://github.com/MKFirmware/MK4duo/commit/c8a8b086c24167d7ec260ca796cd7c71a958bdb6

closed 2018-08-14 by iosonopersia

wesk-ms, 2018-08-14:
@iosonopersia Thank You!!

commit c8a8b086 (2018-08-14): Fix #558