MKFirmware/MK4duo #565: SPI INITIAL RATE

Micilera, 2018-08-20:
Firmware version v4.3.6
We are using an Arduino due, HW SPI for the SD Card on a:
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

In ..\src\sd\SDFat.h line 440 is defined:

  #define SPI_SD_INIT_RATE 11

But in ..\src\HAL\HAL_DUE\HAL_spi_Due.cpp, line 153, there is an array:

Micilera, 2018-08-20:
// 8.4 MHz, 4 MHz, 2 MHz, 1 MHz, 0.5 MHz, 0.329 MHz, 0.329 MHz
  int spiDueDividors[] = { 10, 21, 42, 84, 168, 255, 255 };

of only 7 entries.

Thanks.
Micilera

GuttalBug, 2018-08-20:
More than a "Compiler error" this sound as a conceptual error

iosonopersia, 2018-08-25:
@Micilera , @MagoKimbra , what do you think of these changes? https://github.com/iosonopersia/MK4duo/commit/3b9f9c972d06d57ec23046e65aadf2e6989c9939

Micilera, 2018-08-29:
iosonopersia,
I see a lot of changes but I think that is only mandatory to change the default spi rate from the index 11 (SPI_SD_INIT_RATE) to another index which is in the boundaries of the spi rate table. Very simple.

MagoKimbra, 2018-08-31:
On the two, SPI is initialized at the beginning and is no longer done then the HAL :: spiInit routine in sdfat is not executed.
HAL :: spiBegin () performs SPI initialization with spiInit (1);
I modify this so that it initializes with spiInit (SPI_SPEED);
SPI_SPEED depends on whether you have put SDSLOW or SDEXTRASLOW.

iosonopersia, 2018-09-01:
I don't know if it helps, but this is a list of all the locations in code where `spiBegin` and `spiInit` are called: [SPI.txt](https://github.com/MKFirmware/MK4duo/files/2342180/SPI.txt)

Actually, this issue was solved but we will keep it opened to collect ideas on how to clean up SPI code.

closed 2018-09-04 by iosonopersia

iosonopersia, 2018-09-06:
See #576 to find an explanation of the SPI connection management across different platforms in MK4duo.