MKFirmware/MK4duo #566: SPI RATE

Micilera, 2018-08-20:
Firmware version v4.3.6
HW SPI on an arduino DUE using an SDCard

In ..\src\HAL\HAL_DUE\HAL_spi_Due.cpp, there are two member functions: HAL::spiBegin() and HAL::spiInit().
Inside spiBegin there is a call to spiInit(1).

ISSUE 1 -- Why is used a default rate of 4 Mhz (see spiDueDividors[]) in such a fix way?

If the spiBegin is called once, the flag spiInitMaded is set to true. So, after this call all the following calls to spiInit() do nothing. 
In this way in Sd2Card::init(), line 3142 spiInit() does nothing because spiBegin is called few line above.
More than that, if the init function return OK, so that the final rate would be set, the call of  spiInit() in every chipselect() calls does nothing. 

ISSUE 2 -- At the end, the SDCard runs at the rate hard coded  in spiBegin().

In ..src\HAL\HAL_spi_Due.cpp, line 166:

      if (spiRate > 6) spiRate = 1;

ISSUE 3 -- Why is made that choice? It would be more reasonable to select as the default rate the slowest, not one of the faster.

Thanks.

iosonopersia, 2018-08-25:
@Micilera , @MagoKimbra Maybe we should let the user choose the SPI rate... while we should check more carefully SDCard code...

MagoKimbra, 2018-08-25:
The sdcard code is the standard for arduino that is fine if you make a program that reads an sd, the fact that there is not only the peripheral, then it must be initialized immediately at the beginning and then that of the sdcard in the end no longer needed. .
I have already made a nice clean from the original Marlin code, needs further cleaning, but then you have to try and try again ...
Now until the end of the month I have no time, I'm out with my daughter, let's see what you can do when I come back, if in the meantime you want to play with it and try it out ...

iosonopersia, 2018-09-01:
@Micilera `ISSUE 1` and `ISSUE 2` are now fixed. What it's left is `ISSUE 3`.

The default behaviour is max speed for SDCard connection, since slower speeds should be explicitly selected by the user enabling `SDSLOW` or `SDEXTRASLOW` in the configuration. One of the advantages of the SDCard over the USB+host option is its speed and reliability, so that's what users should expect by default.

Anyway, actually `spiRate = 1` corresponds to the `SDSLOW` option, while the fastest speed would be achieved through `spiRate = 0` . `SDSLOW` is half the full speed, so that's should be a safe value.

I'm going to close this issue but I recognise there's still some work to do for SPI.
Thank's a lot @Micilera for helping us!

closed 2018-09-01 by iosonopersia

MagoKimbra, 2018-09-01:
The problem is that the spi port can not be initialized differently, so in theory it must go to the slower peripheral speed.
When I started MK4duo having the alligator that has the eeprom SPI the maximum reachable value was with SPI_SPEED to 1 that is the divisor 21 with frequency of 4Mhz that's why it was set to 1 by default. Now with the new change I made I have to set SDSLOW to have that value otherwise the eeprom stops at full speed. Then there are the LCD devices or thermocouples.
The problem is how to easily make sure that the user understands, set SPI_SPEED directly and not the speed of the SD? Or??

iosonopersia, 2018-09-06:
See #576 to find an explanation of the SPI connection management across different platforms in MK4duo.