MKFirmware/MK4duo #569: SD and MAX31855

GuttalBug, 2018-08-23:
On MK4DUO version 4.3.6 if both SD and Max31855 are enabled the SD stop to works

iosonopersia, 2018-08-23:
What board did you select? What electronics do you have? Board, display, 8/32 bit ...

GuttalBug, 2018-08-24:
I have Arduino Due, Ramps FD and RepRapDiscount full graphic smart controller

GuttalBug, 2018-08-24:
Yes is correct. MAX31855 is connected to pin 52 for cs, and pin 4 is used for SD cs. Both the device share MISO and MOSI pin.

iosonopersia, 2018-08-24:
@MagoKimbra , max31855 and max6675 thermocouples are SPI devices, right? Their CS pin should be put HIGH immediately at the boot, in the Printer::setup() functions, but that is not being done... Could this be the cause of this issue?

iosonopersia, 2018-08-24:
@MagoKimbra ops, I was wrong. That is being done in Printer::setup_pinout()...

iosonopersia, 2018-08-25:
Maybe this is related to #565 and #566 ...

GuttalBug, 2018-08-28:
Examining the firmware I seem to notice that the MAX31855 is read in interrupt while the SD is read in the main. If this is true this could create the problem because the cs of the MAX can be activated at the same time as the CS of the SD. Surely I'm wrong because it seems strange that such a problem has never been revealed.

closed 2018-09-01 by MagoKimbra

iosonopersia, 2018-09-01:
Fixed by @MagoKimbra

commit e7bc6fd3 (2018-09-01): Fix #569