MKFirmware/MK4duo #576: NEW SPI RATE see also 566

Micilera, 2018-09-01:
Issue #566 is close but the problem is NOT fix at all.

In ..\MK4duo\src\sd\SDFat.cpp, line 3138 there is the following piece of code:

LINE 3138  HAL::spiBegin();
LINE 3139
LINE 3140  // set SCK rate for initialization commands
LINE 3141  spiRate_ = SPI_SD_INIT_RATE;
LINE 3142  HAL::spiInit(spiRate_);

and in ..\MK4duo\src\platform\HAL_DUE\HAL_spi.cpp there is:

  int spiDueDividors[] = { 10, 21, 42, 84, 168, 255, 255 };
  static bool spiInitMaded = false;

  void HAL::spiBegin() {
    if (!spiInitMaded) {
      SPI.begin();
      spiInit(SPI_SPEED);
      spiInitMaded = true;
    }
  }

  void HAL::spiInit(uint8_t spiRate) {
    if (spiInitMaded == false) {
      if (spiRate > 6) spiRate = 1;
      ......
    }
  }

I think it is not very difficult to understand that once the spiBegin() function is called, 
the flag is set and all the following calls to spiInit() have no outcome because the flag is true.
For example the call made in LINE 3142 have no outcome.

It should be clear, but  it is not, that during the Sd2Card::init() the SPI rate is first set very low (see line 3141) and later if it returns OK, the SPI rate is set to what the user requested. But using this flag
the HAL::spiInit() has an outcome just ONE time.

Please fix this bug.

Micilera, 2018-09-04:
Hi,
I have understood that at Sd2Card::init() time, the SPI RATE is low and at the end of the Sd2Card::init() procedure the SPI RATE is set again with the speed requested by the user. If this is NOT true for the DUE platform please let me know and please explain me how it works. This flag (spiInitMaded) in the spiBegin/spiInit of the standard arduino's library is not there. So, my understanding is that LINE 3141/3142 must be executed on the DUE platform as well as on all the other platforms because this is an issue of the SD card management and not of the processor.

iosonopersia, 2018-09-04:
@Micilera @MagoKimbra 
I'll try to reconstruct the correct chronological order of the function calls with DUE:

- MK4duo.ino: `setup` calls `Printer::setup`
- printer.cpp:  `setup` calls, in this order:

  1. `lcd_init`, which is a macro defined in ultralcd.h:27 -> LCD initialized and `HAL::spiBegin` called in u8g_com_shared_hw_spi.cpp:100
  2. `card.mount`, defined in cardreader.cpp:122 -> `fat.begin(SDSS, SPI_SPEED)`, defined in SDFat.cpp:83 -> `SD2Card::init`, defined in SDFat.cpp:3125 -> and finally `HAL::spiBegin` is called
  3. `thermalManager.init` -> if a SPI thermocouple is used, `HAL::spiBegin` will be called at every reading of the max6675 value. I think this can be deleted since the same is NOT done for max31855, which is SPI too.
  4. `externaldac.begin`, defined in externaldac.cpp:37 -> `HAL::spiBegin` is called if you have external Digital to Analog Converter with ALLIGATOR_BOARD

MY SUGGESTION
I think we should define a macro inside Conditionals_post.h like this:
```cpp
#define HAS_SPI_DEVICES  HAS_LCD || HAS_SDSUPPORT || HAS_MAX6675_SS || HAS_MAX31855_SS0/1/2/3 || MB(ALLIGATOR_R2) || MB(ALLIGATOR_R3)
```
And then, write inside `Printer::setup` , at the beginning:
```cpp
#if HAS_SPI_DEVICES
  HAL::spiBegin();
#endif
```

Finally, we should delete every other occurrence of `HAL::spiBegin()` and `HAL::spiInit()` (which would then become useless), just to avoid this great confusion that we have now.

Could this be the first step to clean this up?

iosonopersia, 2018-09-04:
I don't know if it helps, but this is a list of all the locations in code where `spiBegin` and `spiInit` are called: 
[SPI.txt](https://github.com/MKFirmware/MK4duo/files/2348649/SPI.txt)

MagoKimbra, 2018-09-04:
Yes on DUE but on AVR???

MagoKimbra, 2018-09-04:
Scrivo in italiano perché vorrei spiegarlo una volta per tutte poi se vuoi traduci... Con AVR l'inizializzazione è diversa della porta SPI, con il due va fatta una sola volta, ecco perché c'è la variabile booleana...
Quindi in qualsiasi momento si faccia e chi la fa non conta l'importante che non viene più fatta.
Invece con l'AVR la cosa è molto diversa è va fatta cosi com'è per via della SD che va più lenta per questo non posso togliere quella parte...
Ora come è adesso funziona perfettamente sul DUE e SD e AVR e SD...

Micilera, 2018-09-05:
SW  v4.3.6
HW: REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER with SD
Description of the HW in \src\boards\1403.h

Thank You very much for your analysis regarding the HAL:spiBegin() really appreciate.
But I would like to analyze the response of our HW and see on it what is happening.
So I changed the HAL::spiBegin() from:
      static void spiBegin();
to
      static void spiBegin(const char *apc_file, const int asi_line);

and I changed all the calls to HAL::spiBegin() as:

  HAL::spiBegin(__FILE__,__LINE__);

and I print in HAL:spiBegin():
  void HAL::spiBegin(const char *apc_file, const int asi_line) {
    if (!spiBeginMaded) {
      SPI.begin();
      spiInit(2);
      spiBeginMaded = true;
    }
    SERIAL_SMV(ECHO, "HAL::spiBegin ", apc_file);
    SERIAL_MV(" Line: ", asi_line);
    SERIAL_EM(")");
  }

and this is the response of my HW. This is NOT an analysis or a search with an editor; 
it is the response of the HW and it is not in discussion.

The HAL::spiBegin() is called twice very soon at bootstrap time. Both inside Sd2Card::init().
The first time is called from ( the number of the line can be not exact):

Echo:HAL::spiBegin ..\src\sd\SDFat.cpp Line: 3138)
SD card ok

and 
Echo:HAL::spiBegin ..\src\sd\SDFat.cpp Line: 3138)
SD card ok

The second call is made because in ..\src\lcd\ultralcd.cpp:lcd_update() Line: 5204 there is a call to card.mount().
Is this second call another undesiderable feature? In my opinion yes, but it is up to you to check.

So, the HAL::spiBegin is called ONLY two times both in Sd2Card::init().

I think that the SD is on a HW SPI bus. The LCD is on a SW SPI bus (see 1403.h):

  #if ENABLED(NEWPANEL)
    #define LCD_PINS_RS         16
    #define LCD_PINS_ENABLE     17
    #define LCD_PINS_D4         23
    #define LCD_PINS_D5         25
    #define LCD_PINS_D6         27
    #define LCD_PINS_D7         29
  #endif

Due to that there are no HAL::spiBegin() calls about my LCD. 
My LCD is initialized in:
Echo:u8g_dev_st7920_128x64_HAL_4x_fn ..\src\lcd\dogm\u8g_dev_st7920_128x64_HAL.cpp Line: 158)

All of the above is NOT in discussion, it is the response of the HW with the project code.
Back to this issue. The main problem is NOT the HAL:spiBegin() that can be called just one time.

The problem is that after the call to spiBegin() I can't call anymore the spiInit() where I try to change the SPI RATE.
And this is exactly what the Sd2Card::init() would like to do  but it can't, due to this flag (spiBeginMaded).
This is the problem.

In your opinion, at the end of Sd2Card::init() there is:
  return setSckRate(sckRateID);
What it does? Where? How?

My suggestion:
Remember the current SPI RATE each time someone set it. If there is an attempt to set it and if the new value is different from the old value, 
execute the HAL:spiInit() in order to change the SPI RATE and update the current setting; do not use spiBeginMaded in spiInit().

I can open another issue if I have a MAX31855 and NO SD declared at all on my HW, who does the spiBegin?

Obviously my focus is on my HW. It is up to you to generalize what I have suggested you, 
because you have a better overview of all the HW platforms.

Sorry for the long description in a not perfect English. I hope you can understand.
Thanks

iosonopersia, 2018-09-06:
@Micilera , @MagoKimbra 
Ok, I think I'm getting closer to a complete understanding of how this works. The confusion comes from the fact that we have the same code for two different platforms (AVR and DUE). Just imagine what a mess would be to support more platforms!!! 😥

_A little premise: what `HAL::spiBegin` and `HAL::spiInit` do is to initialize (or re-initialize at a different speed) the SPI connection._

  - **So, let's start from DUE.**
Arduino DUE is based on the SAM3X8E microcontroller which runs at 84Mhz. It is so quick that we can set the SPI rate (based on the slowest device) once at the beginning and don't change it anymore. This is **intentionally achieved** thanks to the `spiInitMaded` flag (in correct english it would be `spiInitMade,` but that's not a big problem 😉). So if the code was made for the DUE only, than we could have put one (and only one) `HAL::spiBegin` call in setup() and we could have avoided the use of `spiInitMaded`, exactly as I've proposed in a previous post. **So, why don't we do this?**

  - To answer this question, **let's move one to AVR based platforms.**
AVR microcontrollers run at a slower rate, for example ATMega2560 runs at 16Mhz. This means that we cannot afford to use for a certain device a spiRate which is slower than what the device is capable to do. This means, since for example SD is slower than other devices, that we must adapt the speed of SPI connection every time we change the device we're talking with. This is why in the code there's a `HAL::spiBegin` or a `HAL::spiInit` call at the start of every SPI transaction (just like in case of thermocouples, of external DAC for Alligatorboard, and also of `sdcard.mount` function).

`HAL::spiBegin` and `HAL::spiInit` implementation depends on what platform the user has (AVR or DUE):
  - In case of DUE platform, the implementation makes use of `spiInitMade` to avoid the eventuality of changing spiRate at runtime (as I said before).
  - But in case of AVR, the implementation is different (without the "use-only-once" flag), so that the spiRate will be adapted at runtime based on the device active in that moment.

You can find each implementation by looking at `platform/HAL_AVR/spi.cpp` and `platform/HAL_DUE/HAL_spi.cpp`.

So we have multiple `HAL::spiBegin` calls in the code to be smart with AVR, and optimize its use. But, to be compatible also with DUE, @MagoKimbra modified the HAL_DUE/HAL_spi.cpp implementation of HAL::spiBegin and HAL::spiInit in a way that avoids changes of spiRate at runtime (by using the spiInitMaded flag).

I really hope that this can explain the situation better than what we managed to do up to now.

iosonopersia, 2018-09-06:
So, that being said, we can say that everything is OK as it is now. Well, maybe except for this:
>I can open another issue if I have a MAX31855 and NO SD declared at all on my HW, who does the spiBegin?

If what I wrote in my last post is correct, this **may** become a problem in a configuration with:
  - AVR
  - MAX31855
  - other SPI devices which run a different speeds
In this case, the speed used to talk with MAX31855 could be too quick (connection won't work) or too slow (AVR usage not optimized).

It could also become a problem with this configuration:
  - DUE
  - MAX31855
  - no SD, no spi LCD, no MAX6675
This would be a problem because the code would try to read thermocouple's temperature not having initialized the SPI connection before.

Luckily, this can be fixed very easily: we just have to add a `HAL::spiBegin` call before every reading of the MAX31855 temperature (in function `TemperatureSensor::read_max31855`), just like we do for MAX6675 (in function `TemperatureSensor::read_max6675`). What do you think of this, @MagoKimbra ?

Micilera, 2018-09-07:
I have understood your explanation. I have also understood that the initialization for the DUE must be done only once
and this is a due for this platform. That's OK.
Now I would like to change my hat, from the CPU side to the SD Card side.
I am now an SD Card.
I can be connected via SPI to any CPU of the world, slow, fast, superfast, but I have my timing and if you do not
follow strictly my specifications, I may work, I might not work, my twin could work or my twin could not work.
If the timing specifications are guarantee, our family of devices will work perfectly.
At initialization time I require a slower rate, then if the init phase success the spi rate can be increased. I think this is
not a major issue as the SPI can work on a device basis where each device can have its own clock SPI rate.
I also strongly suggest to increase the spi rate after the initialization, so the system will be faster and if I am very fast I like a very
fast clock rate so I can demonstrates all of my performances.
Please do not slow down a superfast CPU waiting for my slow response when I can work faster and the CPU can be happier.
All of the above has nothing to do with the CPU clock speed. This requirement comes from myself. 
If a super fast CPU can't change the SPI rate
for whichever reason, please ask to an expert how it could be possible to do that, because I can't believe that there is no way to change the SPI rate later on.
By the way, have you ever try on a DUE platform to put an oscilloscope and see (because see is beliving) if the SPI RATE can or can't change?
Again, it is not a matter of HAL::spiBegin() call. It is matter of HAL::spiInit() call.

OK, this is my last comment about this issue. I said what I meant to say, I like the CPUs at their full speed, 
I like the devices running at their full speed fullfilling their specifications.

Micilera, 2018-09-17:
@iosonopersia Ok. Never say never, I am again on this issue because you ask me a question. 
I think I already explained what is going on upon the initialization phase of the SD. 
Anyway I write down something using other words.

Let suppose there is only the SD on the SPI bus and the Arduino DUE as a bus master.

During the initialization we must follow a procedure that there is well documented
in the standard. First of all, the SPI rate must be very low because we don't know
yet which kind of device there is on the other side. If we go slowly,
the devices (also the older) will give us an answer. The opposite is seldom true.

This is what the initialization routine does.
We will see that the SPI rate will be set three times just in this routine.
My focus is on the Arduino DUE platform, I don't know the others.
For your convenience here there is part of Sd2Card::init():

 1) pinMode(chipSelectPin_, OUTPUT);
 2) digitalWrite(chipSelectPin_, HIGH);
 3) HAL::spiBegin();
 4) // set SCK rate for initialization commands
 5) spiRate_ = SPI_SD_INIT_RATE;
 6) HAL::spiInit(spiRate_);

 Line (3) is a sort of file open, inside HAL::spiBegin() (..\MK4duo\src\platform\HAL_DUE\HAL_spi.cpp line 156) 
 there is also a call of spiInit() where the SPI rate is set for the very first time:

  void HAL::spiBegin() {
    if (!spiInitMaded) {
      SPI.begin();
      spiInit(SPI_SPEED);
      spiInitMaded = true;
    }
  }

 Now the flag 'spiInitMaded' is TRUE.

 Due to a recent modification the initial speed is set to 
 'SPI_SPEED', once it was: spiInit(1), ie 4Mhz:

  // 8.4 MHz, 4 MHz, 2 MHz, 1 MHz, 0.5 MHz, 0.329 MHz, 0.329 MHz
  int spiDueDividors[] = { 10, 21, 42, 84, 168, 255, 255 };

Then in line (6) there is an attempt to set it a very low value at SPI_SD_INIT_RATE.
By the way, please carefully read the comment on line (4).

Unfortunately, SPI_SD_INIT_RATE is equal to 11, ie out of bounds of spiDueDividors[], 
but this story is already in another issue.

Back to the attempt to set the rate to SPI_SD_INIT_RATE in line (6), whichever it is,
it will fail (it does nothing) because the flag 'spiInitMaded' is TRUE and the spiInit() looks like:

  void HAL::spiInit(uint8_t spiRate) {
    if (spiInitMaded == false) {
      if (spiRate > 6) spiRate = 1;
      .....
    }
  }

So this second attempt fails. 
Suppose that we have a fast SD card, so the Init() routine ends successfully. 
This is what will happen at the end:

  }
  chipDeselect();
  return setSckRate(sckRateID);

Where setSckRate() just set spiRate_ to the rate set by the user. Who will now set this rate?
Every time there is an attempt to select the SD interface using its chip select:

void Sd2Card::chipSelect() {
  HAL::spiInit(spiRate_);
  HAL::digitalWrite(chipSelectPin_, LOW);
}

there is call to the HAL::spiInit(). All of these attemps will fail due to the flag 'spiInitMaded'.

Three attempts, only the first (in the spiBegin(), so it is a little bit hidden) does something.
I hope I explained the reason why the SW does not change the SPI rate after the first attempt.

End of the story.
Thanks for reading.

iosonopersia, 2018-09-17:
Ok, @Micilera, thanks for the further explanation. **Unfortunately, that is NOT the problem** 😟. I'll try to make it clear. We know very well, because of how the code is written, that SPI rate will not be changed at runtime by the software. **We are perfectly aware of that**. You pointed this out some days ago, and then there was maybe a misunderstanding. Actually @MagoKimbra has always tried to tell you that he knows the code behaves like that (in a way which you're suggesting not to be correct). _**He programmed the code to behave like that**_. He, **for some reason that I don't know** (since I also feel that this behaviour is not 100% ok), chose to program those functions of the DUE HAL in a way which prevents the SPI rate to be changed. I hope that we won't have to explain this anymore. I'm ready to apologize if I wasn't this clear before.

**Now**, I'm not asking why it behaves like that (_it behaves like that because it's made to behave like that, essentially_). I'm asking why @MagoKimbra chose to program those functions in such a way.
### The question is: why does he think that the current behaviour is correct?
Only if he gives us an answer to this question, than we can discuss to understand if it is ok, otherwise we'll keep talking about nothing...

MagoKimbra, 2018-09-17:
In english i do not write very well, i write in italian and Simone please translate this if you want.. Thank's.
Nel due una volta abilitato il registro della porta spi non può essere cambiata la frequenza di clock se non con un forte delay oppure spegnendo la porta e poi riaprirla. Capisci che non è una cosa che si può fare.
La libreria SD fat è quella derivante da AVR che è diverso quindi alcune cose non sono fatte per Arduino due..
Cmq si vediamo la libreria ufficiale arduino SPI.h e SPI.cpp che trovate nella cartella AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\libraries\SPI\src

```
void SPIClass::init() {
	if (initialized)
		return;
	interruptMode = 0;
	interruptSave = 0;
	interruptMask[0] = 0;
	interruptMask[1] = 0;
	interruptMask[2] = 0;
	interruptMask[3] = 0;
	initCb();
	SPI_Configure(spi, id, SPI_MR_MSTR | SPI_MR_PS | SPI_MR_MODFDIS);
	SPI_Enable(spi);
	initialized = true;
}
```

Come vedete anche quella di arduino una volta inizializzato il registro viene posta la variabile initialized a true e non verrà mai più inizializzata la porta SPI...

Nello stesso modo ho fatto io quando ho scritto HAL_DUE. Ci ho provato ma si blocca la cpu..
Poi nulla vieta di provare altre strade per cambiare l'inizializzazione, ma sia marlin che repetier fanno lo stesso quindi credo che sia giusto cosi...

Ora è un pò più chiaro o no??

Per quanto riguarda l'inizializzazione nelle sonde l'ho tolta perché cmq viene sempre inizializzata la porta SPI prima.

Grazie Simone...

iosonopersia, 2018-09-17:
**Ok, case closed!**
_@Micilera I'm going to translate for you what @MagoKimbra said in the last post._

With DUE microcontroller, once the SPI port register has been enabled, the clock frequency can not be changed except if using a very long delay or switching off the port and then reactivating it. You'll understand that this is not something that can be done in practice.

The SD fat library is the official one that was written for AVR, which is a microcontroller in which the SPI rate can be changed with no problems. So I had to adapt it so that with DUE it wouldn't change SPI rates at runtime...

Anyway we can take a look at the official Arduino library (`SPI.h` and `SPI.cpp` that you can find in the `AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\libraries\SPI\src` folder):

``` cpp
void SPIClass :: init () {
  if (initialized)
    return;
  interruptMode = 0;
  interruptSave = 0;
  interruptMask [0] = 0;
  interruptMask [1] = 0;
  interruptMask [2] = 0;
  interruptMask [3] = 0;
  initCb ();
  SPI_Configure (spi, id, SPI_MR_MSTR | SPI_MR_PS | SPI_MR_MODFDIS);
  SPI_Enable (spi);
  initialized = true;
}
```
As you can see also in the official Arduino library, once the register gets initialized, the `initialized` variable is set to true and the SPI port will never be initialized again...

So I did the same when I wrote HAL_DUE. I tried to change SPI rates dinamically but every time the microcontroller crashed ..
Obviously nothing prevents you to try other ways to change the initialization, but both Marlin and Repetier do the same, so I think this it's the right way ...

Is it now a little clearer or not?

Talking about the initialization of the thermocouples, I removed it because anyway the SPI port is always initialized first.

closed 2018-09-17 by iosonopersia