MKFirmware/MK4duo #579: SD and MAX31855 (issue 569)

GuttalBug, 2018-09-04:
Issue  569 has been closed but is not yet completely fixed. The reading of MAX31855 have been moved from interrupt to main and this solve the conflict with sd but there are still two problems on MAX31855 reading.
1)	On file temperature.cpp line 192 there is a loop that read analog input pin for heathers but if the sensor type of the heather is MAX31855 the only data that can be read is garbage.
2)	The read of MAX31855 performed when the processor is idle not guaranty that the chip have enough time for perform the correction as two consecutive read can be very close together, is necessary wait a time between the read operation like what is implemented for MAX6675

MagoKimbra, 2018-09-04:
Please test now!!!

closed 2018-09-04 by MagoKimbra

iosonopersia, 2018-09-05:
Problem 2) fixed. Fixing problem 1) in this issue: #580

commit 799b3ee1 (2018-09-04): Fix #579

Fix read temp with MAX31855.