MKFirmware/MK4duo #580: SD and MAX31855 (continuation of issue569 and 579)

GuttalBug, 2018-09-05:
The issue 579 parent of 569 have been solved only for what concern the timing of MAX31855 (anyway I think that is not necessary one read cycle every 250uS when this data is used every 100mS, a read every 10mS is more then enough ). The analog read was not taken on consideration. On module temperature.cpp , function set_current_temp_raw on line 192 there is the following statements:
  #if ANALOG_INPUTS > 0
    LOOP_HEATER() heaters[h].sensor.raw = HAL::AnalogInputValues[heaters[h].sensor.pin];
  #endif
But if heater use the MAX31855 what can be read from a analog pin ??? defined where ?
Is necessary a test on sensor type

iosonopersia, 2018-09-05:
Yep, you're right!

iosonopersia, 2018-09-05:
@MagoKimbra , just for curiosity and to learn more: why the sensor pin has to be between 0 and 15? A thermocouple couldn't be connected to a different pin, let's say 36 (for example)? Is your commit good both for AVR and DUE?

Micilera, 2018-09-05:
Is it no better to test the type of the sensor?

MagoKimbra, 2018-09-05:
The normal analog sensors need analog pins with the ADC, the thermocouples are read via SPI, the sensor pin is actually a CS.
It's okay for both ...