MKFirmware/MK4duo #584: Typo default dummy values for Thermistor 999

gearsawe, 2018-09-07:
**Please, follow this template when opening a new issue: this will help us a lot in the process of helping you.**

in Configuration_Temperature.h
under  Thermistor Type
line states  
* 999 : Dummy Table that ALWAYS reads 100 degC or the temperature defined below. 
but value is set to 25 by default. 
#define DUMMY_THERMISTOR_999_VALUE 25
this is a little odd.
Would expect default to be 100 not 25

### Firmware version
4.3.6 - 06092018

MagoKimbra, 2019-02-18:
Questo non l'ho capito!! Si setta come uno vuole, io di solito metto 998 a 200 e 999 a 60 in modo che posso fare prove su schede senza sensori e riscaldatori...

iosonopersia, 2019-02-19:
Sì, infatti. Gearsawe ci ha solo segnalato che poco sopra c'è scritto: "999 : Dummy Table that ALWAYS reads 100 degC or the temperature defined below.", che significa "999: Sensore virtuale che legge sempre 100°C oppure la temperatura definita di seguito."

Per questo si sarebbe aspettato che di default fosse settato a 100 e non a 25. Se secondo te non è un problema possiamo anche chiudere la issue.

iosonopersia, 2019-02-20:
Fixed with commit: https://github.com/MKFirmware/MK4duo/commit/208d5b573820e7156fbbb6497580b4ab6829c50e

closed 2019-02-20 by iosonopersia