MKFirmware/MK4duo #587: MK4duo Configurator issue

carlonb, 2018-09-08:
### Issue description
Cartesiana Ramps3/4 con Arduino Mega, Flusso canalizzatore 4 estrusori, Doppio motore asse Z.

Attualmente ho installato la vers. 4.3.5 ed uso il flusso canalizzatore a 4 vie (quindi 4 estrusori ed un solo hotend). Funziona tutto grazie all'aiuto del Mago che a suo tempo mi ha dato.
Ora sto utilizzando il configuratore per scaricare l'ultima vers. 4.3.6 caricando un Configuration_Overall.h della vers. 4.3.5 (descritta sopra) ma non riesco a selezionare Extruder 1 come socket per asse Z2 (secondo motore Z).
O meglio, se accetto il default proposto "Z2 stepper socket = Extruder 2" (che secondo me dovrebbe essere Extruder 1) cioè, se non ci clicco sopra, allora tutto ok, viene assegnato l'extruder 1 ma se ci clicco sopra allora viene assegnato Extruder 2 che non va bene.

Allego Configuration_Overall e Configuration_Pins della versione 4.3.5 per me funzionante. 

[Configuration_Overall.zip](https://github.com/MKFirmware/MK4duo/files/2363467/Configuration_Overall.zip)
[Configuration_Pins.zip](https://github.com/MKFirmware/MK4duo/files/2363468/Configuration_Pins.zip)

MagoKimbra, 2018-09-09:
Non riesco a capire cosa succeda con il tuo overall, che effettivamente da Extruder 2 e non 1, ma se si rifa la configurazione da zero va.. Bho qualche dato che manda in pallone il resto... Quindi consiglio rifalla da zero...

carlonb, 2018-09-09:
Confermo che rifacendolo da zero è possibile selezionare l'Extruder 1.
Probabilmente caricando il vecchio configuration_overall si porta dentro del casino…

OK, consideriamo chiuso il caso.
Grazie

iosonopersia, 2018-09-09:
@carlonb ma è la stessa cosa che ti era successa nel 2017! #232 Bah, mistero... 😅

carlonb, 2018-09-10:
@iosonopersia Si è la stessa cosa di un anno fa (me ne ero completamente scordato).
Probabilmente c'è qualche incompatibilità caricando nel configuratore i vecchi Configuration_Overall.h.

Ora, ricaricando il Configuration_Overall.h creato partendo da zero nel configuratore funziona correttamente e mi propone il socket per l'Extrder 1.   Mah...

carlonb, 2018-09-14:
Per favore riaprite questa issue dato che purtroppo il problema esiste ancora, non capisco il mio test precedente dove mi era parso tutto risolto.
Vers. mk4duo 4.3.6.
Cartesiana Ramps3/4 con Arduino Mega, Flusso canalizzatore 4 estrusori, doppio motore asse Z (Z+E1).
Allego il mio config_overall dove caricandolo nel configuratore non mi fa piu' scegliere l'Extruder 1 come secondo asse Z.
[Configuration_Overall.zip](https://github.com/MKFirmware/MK4duo/files/2383965/Configuration_Overall.zip)
Grazie

carlonb, 2018-09-15:
Grande Mago Alberto,
Provato a caricare overall vers. 4.3.5 e vers. 4.3.6 sul configuratore, ebbene…… ora mi fa giustamente scegliere Extruder 1 !!!
Problem Solved !