MKFirmware/MK4duo #591: G0 move is not continous at long moves

burner-, 2018-09-16:
### Issue description
With large printers you can do long moves. Some reason firmware will split these moves to several shorter movers. That will slow down printing. 
With command G0 X40 F15000
I assume to get one continous linear move to X40. But instead I got this https://www.youtube.com/watch?v=DBFeBk0lPwU

### Firmware version
4.3.6

MagoKimbra, 2018-09-16:
Yes, but it does so only if it has no commands after the one given.
In print it does not, because the command buffer is always full!

burner-, 2018-09-16:
But why it does that? And I am not so sure that it wont do that while printing. Because I first notice it when I was printing that it do strange stops at when it do long travels. And then I start test it manually.

MagoKimbra, 2018-09-16:
For how the planning is done, it must always calculate one movement with respect to another and therefore a single movement creates problems ... It breaks in two for this speech, but it does so only if it has no other commands immediately afterwards.

burner-, 2018-09-17:
@iosonopersia Thanks. Actually this rise one more question: If planner should chain and adapt transition of moves together why it do brake and acceleration at transition point even there is no angle at vectors?

burner-, 2018-09-17:
For future reference. z jerk was too low. It caused problems with bed autoleveling.

closed 2018-09-17 by iosonopersia