MKFirmware/MK4duo #592: Settings will not get loaded from SD card

burner-, 2018-09-18:
### Issue description
When I save settings to SD card I get following message:
Send: M500
Recv: Echo:Settings Stored (607 bytes; crc 64350)
Recv: Echo:SD EEPROM bytes written: -1

When loading settings from SD card I got following message:
Send: M501
Recv: Echo:SD EEPROM bytes read: 0
Recv: Echo:EEPROM version mismatch (EEPROM=?? MK4duo=MKV51)
Recv: Echo:Factory Settings Loaded

### Firmware version
MK4duo 4.3.6 
Latest commit 465c3fe

### My setup
Ultratronics with sd card

MagoKimbra, 2018-09-18:
he does not write on the SD.
SD EEPROM bytes written: -1
Try to define SDSLOW

burner-, 2018-09-18:
As additional info  M20,M21, M30 works fine. After playing around looks that with #define SDSLOW I manage to get SD card work. So looks that this is not blocking bug but some improvement to error handling and reporting can be good.

MagoKimbra, 2018-09-18:
sdfat gives me an error in writing, how does the firmware understand the reason?
Defective SD? Full SD? I can only print that no byte has been written ..