MKFirmware/MK4duo #597: Radds + arduino Due tmc2208 uart wiring

weed2all, 2018-09-26:
Does anyone knows how to put my tmc2208 into uart mode using arduino due and radds...any wiring diagram will be very good... I've searched to Google but can't find nothing

weed2all, 2018-10-15:
So nobody has done this?i realy don't want to ruin the radds board!

iosonopersia, 2018-12-04:
@weed2all did you find elsewhere the information you needed? 👍

weed2all, 2018-12-05:
Nope...i dont boughter anymore with this...i see no more work on arduino due + radds(aka dead end)so i decided to close it!

MagoKimbra, 2018-12-05:
It's very simple..
https://components101.com/sites/default/files/component_pin/Arduino-Due-Pinout.png
This pinout arduino due the uart is pin 14-15, 16-17, 18-19
In radds this pin used for:
#define SD_DETECT_PIN              14
#define ORIG_Z_ENABLE_PIN      15
#define ORIG_Y_STEP_PIN           17
#define ORIG_Y_DIR_PIN             16
Only pins 18 and 19 are missing, but with just one uart you do not do anything, you could drive only one driver. Perhaps the choice of those drivers was not the best .. Much better the TMC2130.

weed2all, 2018-12-06:
Hi MagoKimbra...those are hardware uart...i want to use a software uart if possible given the time that a software serial library is available...i thought that is easy to implement to mk firmware to use with...and the reassign the available pins on my radds as a uart for at least 3 drivers...the tmc library maker said that i could try to implement in tmc library and the to do a pull request...the bad thing is that i dont know how to program it...if anyone have the time and will to make it work will be awsome...but in almost 3months seen no activity on this and that was the reason that i closed the issue...
P.S.
I love marlin, and mk4duo but with the money i had spent on radds, tmc2208,and arduino due...it was wisely to buy the duet WiFi board...whitch I will do soon...

MagoKimbra, 2018-12-06:
As I have already explained, it is not MK4duo who must use the serial software library, but the TMC library.
If you open the TMC library you will find this:
`
#define SW_CAPABLE_PLATFORM defined (__ AVR__) || defined (TARGET_LPC1768)
`
And if SW_CAPABLE_PLATFORM is defined, then you can use the TMC2208 with serial software.
#if SW_CAPABLE_PLATFORM
TMC2208Stepper (uint16_t SW_RX_pin, uint16_t SW_TX_pin, float RS, bool has_rx = true);
#endif
This means that for now you can not use the TMC2208 with SAM.
Also because the software library uses the timers to be able to manage it and the timers are used for the steppers and other things and there are no available to manage the serial software.
Now if you want to take the Duet, you can not say no.
I can not do anything else.

MagoKimbra, 2018-12-06:
Writing for the TMC library is not my role, I break the firmware, it takes someone to do this. We do not do anything for Marlin.
The duet is a unique card that was made for the firmware and the firmware was written for the card. This makes everything easier and more efficient.
I repeat that you made the mistake you made by choosing that card and TMC2208 drivers. Many use the Ruramps 1.3 on two arduino with the TMC2130 drivers and it works very well. The SPI port is unique and needs only one pin (any) to make the Chip Select.

Weruminger, 2019-01-11:
I am on doing to program and Test UART multiplexing . It Sems to work (Test UART switching  only) but i can not expect the effort to realize a funktional draft . I am also busy in Business.

Weruminger, 2019-01-11:
Btw you Need additional Hardware. I use a Quad Switch CD4066.

weed2all, 2019-01-14:
Tell me more how do you implement this...i can go buy a pack of CD4066 from aliexpress or maybe find it locally...how many tmc2208 in uart you use?this option is available to use more than 4 uart?and in software what need to be done?

closed 2019-03-20 by weed2all