MKFirmware/MK4duo #598: Activation of more than 1 serial port

Andy-27, 2018-10-02:
### Issue description

If in file "Configuration__Basic.h" the serial port is set like:
52  #define SERIAL_PORT 0 
I can use a PC connected via USB to Arduino Programming Port for controlling the printer, e.g. send G-Codes, etc. I have also conected a MKS 3.2 touch display to serial port 1 on RADDS board. This display is not working then, meaning it will not display actual temperatures nor can control printer by sending out G-Codes.

If I change setting in  file "Configuration__Basic.h" the serial port like this:
52  #define SERIAL_PORT 1
The MKS touch display is working properly, it shows the actual temperatures and i can control printer, e.g. homing. A PC connected to programming port cannot control the printer any longer nor receiving printer information.

For sure the baud rate is set up properly in configuration files as well as in MKS display.

###QUESTION:
Is there a way to activate both serial ports (1 and 0) in parallel?
If so, how to do?

### Firmware version   4.3.6

Looking forward for any idea to get both ports active without patching firmware any time.
Many thanks for your kindest support

Ciao
Andreas

MagoKimbra, 2018-10-02:
Marlin 2.0 have two ports serial in parallel, MK4duo for now not have this feature.

Andy-27, 2018-10-02:
gracie Alberto for the quick response

Is there might be another way. I could have a switch, connetcted to Y-Max endstop input pin (36), where I could change between 1 and 0. In this case I would add a line in "configuration_pins.h": 
#define SWITCH_SERIAL  36
But where exactly would I need to place the retrieve of SWITCH_SERIAL in the "cpp" file(s)?
Could think on a simple multiplication, set serial port always to 1 in configuration_basic.h and then do retrieve by adding "* SWITCH_SERIAL". The result is then "0" if switch is connected to ground or "1" if connected to VCC.   ;-)

iosonopersia, 2018-10-24:
@Andy-27 MK4duo's branch named `4.3.7beta` has now this function. In Configuration_Basics.h you can define `SERIAL_PORT_1`/`BAUDRATE_1` and `SERIAL_PORT_2`/`BAUDRATE_2`.

closed 2018-10-24 by iosonopersia

weed2all, 2018-10-25:
Where i can find the 4.3.7 beta?can you post a link to direct download?

weed2all, 2018-10-25:
Thank you buddie!i will try it as soon as i arrive home

Andy-27, 2018-10-25:
Bon giorno,
I did the download yesterday evening and start seting up all my specific parameters in config files. At compilation, a lot of errors occur. I'll work on it and share with you my findings later.
What I so far dedected was an error message related to \src\feature\bedlevel\math\vector_3.h refering to last comand in file #endif in row 83 saying ... #endif without #if before.
Where in prior version was the beginning of this file with #ifndef ......  & #define ....., it is now saying #pragma once
But this is just one error I dived into. Will come back with other findings.

Ciao
Andreas

weed2all, 2018-10-25:
Thats cool man.. thank you...when you have time as well and if you can send the diagram wiring for mks tft 32 to radds board...i would like to see how perform the mks lcd over nextion lcd

Andy-27, 2018-10-25:
Hi weed2all,
I did it very simple and it works. I used the serial port 1 which is on AUX1 pins on right RADDS side, staring with +5V, TX1, RX1, GND...... Do not forget to cross the connections, from RADDS TX to MKS display RX and wise versa. This RADDS TX to MKS RX I did straight away w/o any level shifter since the 3V from DUE is understood by MKS as HIGH. The MKS display TX to RADDS RX I did with a resistor of 4,7k from MKS TX to RADDS RX and another resistor 10k from RADDS RX to GND. As simple as that. In former 4.3.6 version I set SERIAL_PORT to 1 and MKS display shows e.g. temperatures of sensors and also allowing e.g. homing by pushing icon on touch display.

Hope that helps, may play also a bit with resistor values to lower down the 5V signal from MKS to 3,3V for DUE.

weed2all, 2018-10-25:
Ohh...is just like nextion...uses only 4 wires...thats nice...thank you alot...was made me some confusion because the mks tft has more than 4 wires...

Andy-27, 2018-10-25:
I need to correct myself in regards to the resistor network for TX from MKS display to DUE RX. It must be 39k and 82k to GND. With the 10k and 4.7k mentioned above the HIGH signal shrinks too much.

weed2all, 2018-10-25:
So in order of things...to connect the mks tft to radds board I will need a 39k resistor from mks TX to radds RX and on radds board need to jumper the RX pin to GND with a 82k resistor?

Andy-27, 2018-10-25:
correct - I couldn't agree more   ;-)

Andy-27, 2018-10-25:
with this set-up the high signal from MKS TFT (5V) will be 3.3V at DUE/RADDS

weed2all, 2018-10-25:
Awsome...thank you buddie...will test it asap!