MKFirmware/MK4duo #602: MACRO ZERO and compiler warning

Micilera, 2018-10-10:
v4.3.6 
All configuration.
The compiler is complaining about the call to the macro in ..\MK4duo\src\core\planner\planner.cpp line 2284. 
As the argument of the macro is a pointer, the result is not what is expected.

iosonopersia, 2018-10-24:
@Micilera Do you think this may lead to a dangerous bug? We should check this...

Micilera, 2018-10-24:
This is not a 'suggestion', this is a bug at least in what you mean to do. If p is a pointer, sizeof(p) and sizeof(*p) have different meaning. if p is a pointer to a structure,  ZERO(p) does not clean the structure as ZERO is defined as:

#define ZERO(a)           memset(a,0,sizeof(a)) 

Hope the above can help

Micilera, 2018-10-25:
No no no. This will be another bug!
I would propose to use directly the library function instead of using the macro:

memset(block,0,sizeof(block_t));

MagoKimbra, 2018-10-25:
Thank's Micilera you are right!!

closed 2018-10-25 by MagoKimbra