MKFirmware/MK4duo #604: Bltouch smart v2 genuine problem

robertoscipione, 2018-10-16:
### Issue description
Ciao, sto trovando problemi nell'installazione del bltouch smart v2, ho letto vari post su reprap seguiti dal mago ma niente non ne vuole sapere. Si accende solo il led rosso e non il blu per il segnale del servo inoltre il comando m280 mi ritorna con unknow command.

Su marlin funziona quindi sara sicuramente un errore da parte mia di configurazione.

Chiedo gentilmente se e possibile scrivere in questo post la procedura corretta passo passo per il funzionamento del sensore in mk4duo.

Grazie in anticipo per l'aiuto.

### Firmware version
4.3.6

### My setup
Cartesiana

Complimenti per fw veramente un bel lavoro anche se lo sto studiando da poco, sono sempre stato un marliniano, ma sono italiano ;)

MagoKimbra, 2018-10-16:
Hai attivato i servo nella configurazione? Hai messo che hai un servo? Se ti dice che unknow al comando M280 significa che non è attivo nessun servo...

robertoscipione, 2018-10-16:
Ok mia disattenzione, usando spesso marlin andavo a configurare i file originali di configurazione e non il file Configuration_Overall.h ( brutte abitudini :) )

Ora funziona il tutto spero che possa essere utile, subito sotto il file con abilitato il supporto al sensore Bltouch Smart V2

Tutte mod effettuate per abilitare il sensore:

in Configuration_Overall.h

1. #define BLTOUCH
2. #define BLTOUCH_DELAY 375
3. #define ENABLE_SERVOS
4. #define NUM_SERVOS 1

in Configuration_Pins.h

1. #define Z_PROBE_PIN ORIG_Z_MIN_PIN

###################################
Configuration_Overall.h
###################################

```
/**
 * MK4duo Firmware for 3D Printer, Laser and CNC
 *
 * Based on Marlin, Sprinter and grbl
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 * Copyright (C) 2013 Alberto Cotronei @MagoKimbra
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program. If not, see <http://www.gnu.org/licenses/>.
 */

/**
 * Configuration_Overall.h
 * Here you can define all your custom settings and they will overwrite configurations in the main configuration files.
 */

/*******************************
 *   Firmware Version V4.3.6   *
 *******************************/

#define CONFIGURATION_OVERALL

/***********************
 * Configuration_Basic *
 ***********************/
#define SERIAL_PORT 0
#define BAUDRATE 250000
#define STRING_CONFIG_H_AUTHOR "(none, default config)"
#define MACHINE_UUID "6e8f83e2-c930-0417-c855-67227dc35639"
#define KILL_METHOD 0
#define NO_TIMEOUTS 1000
//#define ADVANCED_OK
//#define EMERGENCY_PARSER
//#define FASTER_GCODE_PARSER
//#define FASTER_GCODE_EXECUTE
#define HOST_KEEPALIVE_FEATURE
#define DEFAULT_KEEPALIVE_INTERVAL 2
#define MOTHERBOARD BOARD_RAMPS_13_HFB
[... 1796 lines cut ...]
  #define RGB_LED_W_PIN NoPin
#endif

#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_PIN NoPin
#endif

#if ENABLED(DHT_SENSOR)
  #define DHT_DATA_PIN NoPin
#endif

#if HAVE_DRV(TMC2130) && ENABLED(SOFT_SPI_TMC2130)
  #define SOFT_MOSI_PIN 51
  #define SOFT_MISO_PIN 50
  #define SOFT_SCK_PIN  52
#endif

//============================================================================

#endif /* _CONFIGURATION_PINS_H_ */
```

Grazie per ora, mi scuso per aver disturbato inutilmente ;)

Buon lavoro

MagoKimbra, 2018-10-16:
Ma scusami la modifica non potevi farla fare al configuratore?? Importavi il tuo overall assegnavi il servo e riscaricavi l'overall bello e pronto con il firmware..

robertoscipione, 2018-10-16:
Grazie per il consiglio, ma tutto quello che è auto non mi gusta, preferisco il codice, almeno capisco realmente quello che faccio fare alla mia 3d printer.

Vecchia scuola :),
Grazie Mago ;)

MagoKimbra, 2018-10-16:
Ok.. Come preferisci, io faccio tutto a mano...

closed 2018-10-16 by MagoKimbra

robertoscipione, 2018-10-16:
Naturalmente lo hai scritto tu :)

MagoKimbra, 2018-10-16:
Cosa??
Intendevo dire che anche io faccio a mano, la configurazione, per provare a compilare con diverse impostazioni e vedere se da errore...

robertoscipione, 2018-10-16:
Si intendevo esattamente quello che hai detto

robertoscipione, 2018-10-16:
Cmq volevo chiederti dato che sono quasi alla fine del mio progetto, ho
ridisegnato una prusa i3 e la rilascero open se non vi dispiace la volevo
rendere pubblica con l'mk4duo e citarvi, la cosa bella che e la prima prusa
i3 completamente italiana 😄😄😄

MagoKimbra, 2018-10-16:
Se citi mi fa solo che piacere... Ma non è la prima prusa I3 completamente Italiana... C'è la I3 EVO da almeno 3 anni...

robertoscipione, 2018-10-16:
Si conosco ma guarda la foto e capirai

MagoKimbra, 2018-10-16:
Che foto??

robertoscipione, 2018-10-16:
[image]

robertoscipione, 2018-10-16:
[image]

robertoscipione, 2018-10-16:
il piatto lavora molto bene ed è stabile, spool integrato con motore, supporto estrusore per stampare tutti i materiali integrato bltouch e naturalmente la camera che così progettata la smonti in 5 min ed è utilizzabile con qualsiasi stampante basta allungare i rod e tagliare il policarbonato in base alle proprie esigenza

robertoscipione, 2018-10-16:
che ne pensi ?

robertoscipione, 2018-10-16:
Cmq la revo e closed not open solo stl no sorgenti