MKFirmware/MK4duo #607: New Version 4.3.7 - Compiler Errors

Andy-27, 2018-10-25:
### Issue description
I have compilation errors with the new beta version 4.3.7 

### Compile errors
I have attached 2 files, one showing the errors when SERIAL_PORT_2 is switched off (-2) and one when it is set to port 1. SERIAL_PORT_1 is still set to -1

### Firmware version
MK4duo's version BETA 4.3.7
Arduino IDE 1.8.7

### My setup
I just did the settings in all config files as used in version 4.3.6

[Andy-27 Errors with SERIAL_PORT_2 -2.txt](https://github.com/MKFirmware/MK4duo/files/2516436/Andy-27.Errors.with.SERIAL_PORT_2.-2.txt)
[Andy-27 Errors with SERIAL_PORT_2 1.txt](https://github.com/MKFirmware/MK4duo/files/2516437/Andy-27.Errors.with.SERIAL_PORT_2.1.txt)

Would appreciate a lot your kindest support to solve the issue.

Grazie mille

MagoKimbra, 2018-10-26:
Please attach your configuration files... This error is very strange, possible is arduino ide problem...

Andy-27, 2018-10-26:
please find attached my config files- only those I changed.
[Configuration_Basic.h.txt](https://github.com/MKFirmware/MK4duo/files/2518570/Configuration_Basic.h.txt)
[Configuration_Temperature.h.txt](https://github.com/MKFirmware/MK4duo/files/2518571/Configuration_Temperature.h.txt)
[Configuration_Feature.h.txt](https://github.com/MKFirmware/MK4duo/files/2518572/Configuration_Feature.h.txt)
[Configuration_Core.h.txt](https://github.com/MKFirmware/MK4duo/files/2518573/Configuration_Core.h.txt)
[Configuration_Pins.h.txt](https://github.com/MKFirmware/MK4duo/files/2518574/Configuration_Pins.h.txt)

iosonopersia, 2018-10-28:
I've just fixed the errors when using `SERIAL_PORT_2 -2`. I'm able to compile it with Arduino IDE 1.9.0-beta.

As for the other errors, they all come from file `m500_m503.h`: the problem is that gcode_t struct does not contain a field called "port". Please @MagoKimbra check that file...

Andy-27, 2018-10-28:
That sound's good, grazie mille

iosonopersia, 2018-10-29:
The second error was solved here: https://github.com/MKFirmware/MK4duo/commit/003f1cd015c11ee847257a2c3f239526106ae8ac

closed 2018-10-29 by iosonopersia

Andy-27, 2018-10-31:
Gentlemen, I am very surprised. The corrections you made are working well. I tested the function on 2nd serial port last night and everything went perfect.
Thank you so much for the outstanding work you did.
During the trial I made, i found some small things which I'll pos in new issue request.
Grazie mille