MKFirmware/MK4duo #612: 4.3.7 won't compile when using tmc2208 drivers.

wesk-ms, 2018-10-30:
**Please, follow this template when opening a new issue: this will help us a lot in the process of helping you.**

### Issue description
Tell us here what happened: don't miss any valuable detail, be clear and concise, don't use UPPERCASE or abbreviations which can make understanding what you wrote quite difficult.

### Compile errors
If you have problems compiling MK4duo, report here the error messages Arduino IDE gives you.

Before posting this issue, however, take a long breath and check this two things:
1. If you're not using the latest version of Arduino IDE, please try again with the latest one.
2. Always check if an error (or more than one) starts with `DEPENDENCY ERROR: _something wrong!_`: **DO NOT POST DEPENDENCY ERRORS**, please! They are messages which tell you what config option was set badly. Fix your configuration and try again. If you also get other types of error, please be sure to have fixed dependency errors **before** posting this issue!

### Firmware version
MK4duo's versions that have this issue.

### My setup
If you used the online configurator, copy the content of `Configuration_Overall.h` and `Configuration_Pins.h` in two text files. **Then Drag&Drop them here. DO NOT COPY THE CONTENT DIRECTLY HERE!!!**

Have you configured MK4duo manually? Then write here a list of the settings that may be related to the issue:
- SETTING_CHANGED_1    true
- SETTING_CHANGED_2    25
- SETTING_CHANGED_3    "blablabla"
- ...

_If needed, you'll be asked for other settings you modified._

**Please be patient, we'll be trying to help you as soon as possible.**

Just getting this bug opened while I'm thinking about this issue.  Will add the details on the many compile errors when I get back to my PC.

Easy to repro, though.  Just specify a tmc2208 driver for at least 1 axis being used and try to compile.

wesk-ms, 2018-10-31:
Adding compile errors:

In file included from sketch\MK4duo.h:151:0,

                 from c:\MK\MK4duo\MK4duo.ino:318:

sketch\src/feature/tmc/tmc.h: In constructor 'MKTMC::MKTMC(char*, uint8_t, uint16_t, uint16_t, float, bool)':

sketch\src/feature/tmc/tmc.h:193:21: error: no matching function for call to 'TMC2208Stepper::TMC2208Stepper(uint16_t&, uint16_t&, float&, bool&)'

         id(DRIVER_ID)

                     ^

sketch\src/feature/tmc/tmc.h:193:21: note: candidates are:

In file included from sketch\src/feature/tmc/tmc.h:34:0,

                 from sketch\MK4duo.h:151,

                 from c:\MK\MK4duo\MK4duo.ino:318:

C:\Users\user\Documents\Arduino\libraries\TMCStepper\src/TMCStepper.h:645:3: note: TMC2208Stepper::TMC2208Stepper(Stream*, float, bool)

   TMC2208Stepper(Stream * SerialPort, float RS, bool has_rx=true);

   ^

C:\Users\user\Documents\Arduino\libraries\TMCStepper\src/TMCStepper.h:645:3: note:   candidate expects 3 arguments, 4 provided

C:\Users\user\Documents\Arduino\libraries\TMCStepper\src/TMCStepper.h:642:7: note: constexpr TMC2208Stepper::TMC2208Stepper(const TMC2208Stepper&)

 class TMC2208Stepper : public TMCStepper {

       ^

C:\Users\user\Documents\Arduino\libraries\TMCStepper\src/TMCStepper.h:642:7: note:   candidate expects 1 argument, 4 provided

C:\Users\user\Documents\Arduino\libraries\TMCStepper\src/TMCStepper.h:642:7: note: constexpr TMC2208Stepper::TMC2208Stepper(TMC2208Stepper&&)

C:\Users\user\Documents\Arduino\libraries\TMCStepper\src/TMCStepper.h:642:7: note:   candidate expects 1 argument, 4 provided

In file included from sketch\src/platform/platform.h:35:0,

                 from sketch\MK4duo.h:111,

                 from c:\MK\MK4duo\MK4duo.ino:318:

sketch\src/feature/tmc/tmc.h: In static member function 'static void TMC_Stepper::get_sgt(MKTMC*)':

sketch\src/feature/tmc/tmc.h:418:47: error: 'class MKTMC' has no member named 'sgt'

       SERIAL_EMV(" homing sensitivity: ", st->sgt(), DEC);

                                               ^

sketch\src/platform/common/communication.h:118:57: note: in definition of macro 'SERIAL_VAL'

 #define SERIAL_VAL(val, ...)                (Com::print(val, ## __VA_ARGS__))

                                                         ^

sketch\src/platform/common/communication.h:138:49: note: in expansion of macro 'SERIAL_MV'

 #define SERIAL_EMV(msg, val, ...)           do{ SERIAL_MV(msg, val, ## __VA_ARGS__); SERIAL_EOL(); }while(0)

                                                 ^

sketch\src/feature/tmc/tmc.h:418:7: note: in expansion of macro 'SERIAL_EMV'

       SERIAL_EMV(" homing sensitivity: ", st->sgt(), DEC);

       ^

In file included from sketch\MK4duo.h:151:0,

                 from c:\MK\MK4duo\MK4duo.ino:318:

sketch\src/feature/tmc/tmc.h: In static member function 'static void TMC_Stepper::set_sgt(MKTMC*, int8_t)':

sketch\src/feature/tmc/tmc.h:422:11: error: 'class MKTMC' has no member named 'sgt'

       st->sgt(sgt_val);

           ^

In file included from sketch\src/platform/platform.h:35:0,

                 from sketch\MK4duo.h:111,

                 from c:\MK\MK4duo\MK4duo.ino:318:

sketch\src/feature/tmc/tmc.h: In static member function 'static void TMC_Stepper::get_disable_I_comparator(MKTMC*)':

sketch\src/feature/tmc/tmc.h:463:49: error: 'class MKTMC' has no member named 'disfdcc'

       SERIAL_EMV(" disable_I_comparator: ", st->disfdcc());

                                                 ^

sketch\src/platform/common/communication.h:118:57: note: in definition of macro 'SERIAL_VAL'

 #define SERIAL_VAL(val, ...)                (Com::print(val, ## __VA_ARGS__))

                                                         ^

sketch\src/platform/common/communication.h:138:49: note: in expansion of macro 'SERIAL_MV'

 #define SERIAL_EMV(msg, val, ...)           do{ SERIAL_MV(msg, val, ## __VA_ARGS__); SERIAL_EOL(); }while(0)

                                                 ^

sketch\src/feature/tmc/tmc.h:463:7: note: in expansion of macro 'SERIAL_EMV'

       SERIAL_EMV(" disable_I_comparator: ", st->disfdcc());

       ^

In file included from sketch\MK4duo.h:151:0,

                 from c:\MK\MK4duo\MK4duo.ino:318:

sketch\src/feature/tmc/tmc.h: In static member function 'static void TMC_Stepper::set_disable_I_comparator(MKTMC*, bool)':

sketch\src/feature/tmc/tmc.h:467:11: error: 'class MKTMC' has no member named 'disfdcc'

       st->disfdcc(onoff);

           ^

In file included from sketch\src/platform/platform.h:35:0,

                 from sketch\MK4duo.h:111,

                 from c:\MK\MK4duo\MK4duo.ino:318:

sketch\src/feature/tmc/tmc.h: In static member function 'static void TMC_Stepper::get_stealth_amplitude(MKTMC*)':

sketch\src/feature/tmc/tmc.h:481:46: error: 'class MKTMC' has no member named 'pwm_ampl'

       SERIAL_EMV(" stealth_amplitude: ", st->pwm_ampl());

                                              ^

sketch\src/platform/common/communication.h:118:57: note: in definition of macro 'SERIAL_VAL'

 #define SERIAL_VAL(val, ...)                (Com::print(val, ## __VA_ARGS__))

                                                         ^

sketch\src/platform/common/communication.h:138:49: note: in expansion of macro 'SERIAL_MV'

 #define SERIAL_EMV(msg, val, ...)           do{ SERIAL_MV(msg, val, ## __VA_ARGS__); SERIAL_EOL(); }while(0)

                                                 ^

sketch\src/feature/tmc/tmc.h:481:7: note: in expansion of macro 'SERIAL_EMV'

       SERIAL_EMV(" stealth_amplitude: ", st->pwm_ampl());

       ^

In file included from sketch\MK4duo.h:151:0,

                 from c:\MK\MK4duo\MK4duo.ino:318:

sketch\src/feature/tmc/tmc.h: In static member function 'static void TMC_Stepper::set_stealth_amplitude(MKTMC*, uint8_t)':

sketch\src/feature/tmc/tmc.h:485:11: error: 'class MKTMC' has no member named 'pwm_ampl'

       st->pwm_ampl(stealth_amplitude_val);

           ^

exit status 1
[Error] Exit with code=1

iosonopersia, 2018-12-04:
Sorry, I put the `Solved issue` label because I thought this was already fixed. Actually, I managed to reproduce this error, so this must be solved.

iosonopersia, 2019-02-07:
Solved by @MagoKimbra with latest commit: https://github.com/MKFirmware/MK4duo/commit/6b83d2cd6dff36e47182ad79eba8e1522d5741c1

closed 2019-02-07 by iosonopersia