MKFirmware/MK4duo #613: New Version 4.3.7 - Findings

Andy-27, 2018-10-31:
### Issue description
On LCD in menue items Configuration and also in Temperatur, there are sub-menues for pre-heating settings for 3 different materials. The first and 2nd entry (PLA and ABS) is showing the menue text in German, while the 3rd entry (GUM) still shows text in English.

### Compile errors
none

### Firmware version
4.3.7 beta   28.10.2018

### My setup
My 3d printer has different print heads. One is with a BL-Touch bed probe which is used for homing, the other don't has this and uses the Z endstop for homing. Whenever you choose a bed-probe in configuration, the firmware will home Z only with the probe as endstop. To be able to toggle between Z homing with Z-endstop or BL-Touch without patching the firmware every time, I implement a switch which is connected to X-MAX and put retrieves in mechanics.cpp and, in my case, core_mechanics.cpp.
I'll attach the corresponding files and an explanation.
May be this could be interesting for you too.

Thanks

Andy-27, 2018-10-31:
[Z-Homing change with switch.txt](https://github.com/MKFirmware/MK4duo/files/2535325/Z-Homing.change.with.switch.txt)
[Configuration_Pins.h.txt](https://github.com/MKFirmware/MK4duo/files/2535326/Configuration_Pins.h.txt)
[core_mechanics.cpp.txt](https://github.com/MKFirmware/MK4duo/files/2535327/core_mechanics.cpp.txt)
[mechanics.cpp.txt](https://github.com/MKFirmware/MK4duo/files/2535328/mechanics.cpp.txt)

iosonopersia, 2019-03-27:
Hi @Andy-27 , sorry for the delay. This issue required an update of the language files. It should be fixed now, try downloading the newest version from the master branch!

closed 2019-03-27 by iosonopersia